FormerLurker/Octolapse #707: Something is wrong with my Timelapse using A6000 Camera

kot0005, 2021-03-14:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.41

#### Version of OctoPrint

OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

Timelapse does not look right, its appears as if only a couple of snapshots were used to render it, however my snapshot count was 75 with no errors.

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: octo pi
Os Version: 0.17

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://youtu.be/QYEPDJxOzOI

So is there a way to download the snapshots to my PC or external storage and render the timelapse on my PC ?

I am using a Sony A6000

Theres no errors , but i dunno why the timelapse looks like this.

kot0005, 2021-03-14:
looks like the rendered timelapse is too fast , even using the slowest speed in youtube seemed like the video was playign really fast.

Is there a way to have the snapshots saved to a USB stick attached to the Pi or even just save them in the camera's sd card so i can edit them manually ?

I found the following post https://community.octoprint.org/t/octolapse-how-to-turn-off-saving-and-rendering-to-the-pi/4341 

but its a bit too technical for me  

how do I go about setting up the Option 1 method detailed by FormerLurker ?

do they have to be done using SSH into the Pi or in octolapse itself ?

My camera's name is Sony Alpha-A6000 (Control)

thanks!

kot0005, 2021-03-14:
and how do I make octolapse work when i am using this setting ?
i cant get teh camera script to work when i use manual.

[image]

kot0005, 2021-03-14:
ok, after doing heaps of research, it looks like the Sony A6000 does not support 

https://github.com/gphoto/gphoto2/issues?q=A6000

so am i unable use both these methods ?

1. https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-DSLR-With-No-Download
2. https://github.com/FormerLurker/Octolapse/wiki/DSLR-with-Deferred-Image-Download

kot0005, 2021-03-14:
Also how do i uninstall Gphoto 2 and all the changes i made to its scripts ?

FormerLurker, 2021-03-15:
It's going to take me a bit of time to go through this :)  With regard to gphoto2, I think the install script mentioned in the guide can do this.  If you installed with apt-get, you can use that to uninstall it.  It doesn't take much space though.

Regarding the other changes, just make sure you edit your sudoers file and remove permissions for gphoto2 if you followed that step.  That's really the only critical change.  No need elevating permissions for an app you won't use!

As for the other things, I'll need some time to look through it, but i'm not sure i can be much help with your cam since mine is much simpler :)

kot0005, 2021-03-16:
> It's going to take me a bit of time to go through this :) With regard to gphoto2, I think the install script mentioned in the guide can do this. If you installed with apt-get, you can use that to [...]

So i have enabled archiving and am manually rendering the timelapses and they are workign great. However I would still like the Pi to not render at all and if possible store images onto an eternal usb stick.

am goign to checkout if the wifi to camera method will store shots directly on camera SD.

So i changed my  Trigger profile from Compatibility to Classic every layer and the print quality dropped by a lot, lots of  oozing!!

is there a good Trigger profile you recommend ? compatability was good but not as good as with no timelapse, i was having a bit of oozing and under extrusion near seams

Also my delay is set to 0 but the there is still quite a bit of wait time compared to when i used a webcam. Is there a way to improve this ?

like during snapshots, the printer stays at its stabilization for quite a bit longer on DSLR than with webcam (webcam is almost instant without any waits)

[image]

FormerLurker, 2021-03-16:
> So i changed my Trigger profile from Compatibility to Classic every layer and the print quality dropped by a lot, lots of oozing!!

The smart triggers are the best for quality, which is why they were added to V0.4 (took well over a year to implement these).

> Also my delay is set to 0 but the there is still quite a bit of wait time compared to when i used a webcam. Is there a way to improve this ?

Saving to SD.  Most of the extra delay comes from downloading the huge DSLR images, which are sometimes MANY megabytes depending on your settings, over a slow USB connection.  DSLR transfer is usually done over USB 2.0, but don't expect the max transfer speed....  The print quality guide in the wiki (see troubleshooting and look for print quality) covers this.

You could check out [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-DSLR-With-No-Download), but I don't think your camera supports this.  It is worth a shot still.

kot0005, 2021-03-16:
> The smart triggers are the best for quality, which is why they were added to V0.4 (took well over a year to implement these).
>
>
> Saving to SD. Most of the extra delay comes from downloading the huge [...]

Ahh thanks for the trigger recommendation! wil try it.

Yeah the DSLR method you linked wont work, already tried it

`gphoto2 --auto-detect --get-config capturetarget` gives me an error, i checked with gPhoto2 on github forums and i was told its the camera.

Thanks why i wana to try the wifi method used by "There oughtta be a way"

closed 2021-04-13 by kot0005