FormerLurker/Octolapse #708: [Request] Support for custom codecs

amolloy, 2021-03-14:
#### If this is a feature request describe it here

It would be nice to be able to specify a custom codec for ffmpeg. For example, I am running OctoPrint on an NVidia Jetson Nano and have a version of ffmpeg installed with hardware accelerated H.264 & HVEC encoding. While you could just add the appropriate codecs to the list (h264_nvmpi and hevc_nvmpi, fwiw), it seems like there will always be some random piece of hardware out there that only one person in the world is using for this purpose with some custom built version of ffmpeg to support it. It would be nice to be able to just choose "Custom Codec" then specify the desired codec, rather than either having the list of supported codecs expand forever or leaving those few who want a particular codec out. Thank you for considering!

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-03-15:
I'm trying to work out a way to allow custom ffmpeg commands, which should cover this.  The big issue is the command structure is complex, so it's hard to produce a one-size-fits-all scheme.  It's not impossible, but it will take me time to complete.  There are other bits I need to send to ffmpeg (filename format, output format, framerate, etc..) so I just need to figure out a way to ensure those parts always work, and allow the user to add various other flags and such.  This change affects a massive amount of code too.  Just explaining why you're not likely to see this in the near term, but I AM working on it :)

amolloy, 2021-03-15:
Awesome, thanks for the info!