FormerLurker/Octolapse #709: OctoLapse producing invalid videos from DSLR

jbienz, 2021-03-14:
#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

I followed the guide to setup DSLR capture. All snapshots were taken and a video was produced, but the video is not playable on Windows.

Here is the log file:

[plugin_octolapse (2).log](https://github.com/FormerLurker/Octolapse/files/6137065/plugin_octolapse.2.log)

#### What should have happened?

The video should have played

#### What happened instead?

800 MB of images was converted to a video file only ~20MB in size. I tried multiple profiles. None of them produced a playable video. Windows 10 vide player says "This item is in a format we don't support." 0xc00d36b4

[image]

#### Operating System running OctoPrint and Octolapse

OS Name: Windows 10
Os Version: 1909 build 18363.1440

#### Printer model & used firmware incl. version

Printer Model: MINI+
Printer Firmware Version: OEM from factory

#### Browser and version of browser, operating system running browser

Browser: Tried with both Chrome and Edge
Browser OS: Windows

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: Not Needed

#### Link to settings.json

Link to settings.json with all passwords removed: [settings.txt](https://github.com/FormerLurker/Octolapse/files/6137079/settings.txt)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  
[plugin_octolapse (2).log](https://github.com/FormerLurker/Octolapse/files/6137080/plugin_octolapse.2.log)

#### Link to octoprint.log

Link to octoprint.log:  
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/6137093/octoprint.log)

#### Link to contents of Javascript console in the browser

Link to javascript console output: Not Needed

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: Included Above

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-03-15:
Try playing it with VLC (might need to download it).  Windows has some file resolution restrictions, and I'm guessing your DSLR image is not a standard video resolution.  

If VLC works, you might try downloading handbreak, and changing the resolution in there and re-rendering.  Or, try uploading to youtube.  It can handle almost anything.

jbienz, 2021-03-18:
Thank you @FormerLurker, you are correct that the video did play in VLC. It appears Windows Movies & TV (and thumbnail generator) do not like that resolution.

We can close this issue. I would like to request an option to have OctoLapse generate standard video sizes regardless of snapshot size. I'll open a new issue for that.

closed 2021-03-18 by jbienz

jbienz, 2021-03-18:
Created #713 to track the request for standard resolutions.