FormerLurker/Octolapse #710: Octolapse rendering fails

hemna, 2021-03-16:
Octolapse is able to take snapshots, but when it tries to render the timelapse it fails

The logfile shows

```
2021-03-16 18:29:42,978 - octolapse.render - ERROR - Rendering Error
Traceback (most recent call last):
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_octolapse/render.py", line 1558, in _render
    self._add_text_overlays()
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_octolapse/render.py", line 1987, in _add_text_overlays
    img = self.add_overlay(img,
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint_octolapse/render.py", line 2028, in add_overlay
    font = ImageFont.truetype(font_path, size=font_size)
  File "/octoprint/plugins/lib/python3.8/site-packages/PIL/ImageFont.py", line 852, in truetype
    return freetype(font)
  File "/octoprint/plugins/lib/python3.8/site-packages/PIL/ImageFont.py", line 849, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/octoprint/plugins/lib/python3.8/site-packages/PIL/ImageFont.py", line 173, in __init__
    freetype_version = parse_version(features.version_module("freetype2"))
  File "/octoprint/plugins/lib/python3.8/site-packages/packaging/version.py", line 57, in parse
    return Version(version)
  File "/octoprint/plugins/lib/python3.8/site-packages/packaging/version.py", line 296, in __init__
    match = self._regex.search(version)
TypeError: expected string or bytes-like object
```

Octolapse 0.4.1

octoprint version Version 1.5.3

Octoprint system info

```
browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 11_2_2) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.86 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1400
env.hardware.ram: 915058688
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 3 Model B Plus Rev 1.3
env.python.pip: 21.0.1
env.python.version: 3.8.8
env.python.virtualenv: false
octoprint.safe_mode: false
octoprint.version: 1.5.3
printer.firmware: Marlin Ver 1.0.2
```

FormerLurker, 2021-03-16:
Looks like this is related to the rendering overlay.  There was a bug I fixed relating to this.  See this zip link you can use to install and test via the plugin manager->Get More->From url... in OctoPrint settings(you should be able to re-render via the 'Failed Renderings' button on the tab after you install and reboot):  ```https://github.com/FormerLurker/Octolapse/archive/8cae0c9d8f9b11d7a55c3d6fb04956931636429c.zip```

hemna, 2021-03-17:
I've narrowed it down to the call to features.version_module("freetype2")
It's returning None, which causes parse_version to puke.

FormerLurker, 2021-03-17:
Ug..  this is some odd python version incompatibility within pillow that I've not seen before.  You'll notice that the error is happening entirely within pillow itself:

```
File "/octoprint/plugins/lib/python3.8/site-packages/PIL/ImageFont.py", line 173, in __init__                                                 │
│    freetype_version = parse_version(features.version_module("freetype2"))    
```

Did you upgrade python somehow after Octolapse was installed?

FormerLurker, 2021-03-17:
Ok, I did a bit of research.  I'm guessing you aren't using OctoPi, right?  It could be that the libfreetype package isn't installed on your system, which means there is no text layout support.  You install that package, remove pillow, then re-add pillow (make sure you use the correct virtual environment!) and see if that helps.

hemna, 2021-03-17:
I am running everything inside a container on my rpi.  

```
╰─   ○ cat docker-compose.yml
version: '3.2'

services:
  octoprint:
    image: octoprint/octoprint
    restart: unless-stopped
    privileged: true
    ports:
      - 80:80
      - 8080:8080
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
      - /dev/video0:/dev/video0
    volumes:
      - /home/waboring/docker/octoprint/octoprint:/octoprint
    # uncomment the lines below to ensure camera streaming is enabled when
    # you add a video device
    environment:
      - ENABLE_MJPG_STREAMER=true
      - MJPG_STREAMER_INPUT=-y -n -r 1280x720
 ```

hemna, 2021-03-17:
I tried installing freetype-py and pillow to fix it, but it didn't.

FormerLurker, 2021-03-17:
You need to install libfreetype-dev via apt-get I believe.

hemna, 2021-04-01:
I tried and that doesn't work :(

FormerLurker, 2021-04-01:
Hmmm.  Maybe the kind people who developed pillow may have an idea.  In any case, I will see what I can find.