FormerLurker/Octolapse #711: Octolapse cannot save snapshots

WatEvsCevs, 2021-03-17:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Added user 'octo' to sudoers file for gphoto (as per the octoprint installation from bash script: https://github.com/Nebari-xx/octoprint_installer) as I am running on an OrangPi Zero 2
2. Created 'take-snapshot.sh' in 'home/octo/.octoprint/scripts'
3. Restarted OrangePi
4. Connected to virtual printer in OctoPrint and enabled test mode in Octolapse

#### What should have happened?
Would save the snapshot in the directory specified in 'take-snapshot.sh'

#### What happened instead?
Returns "The 'DSLR - Snapshot Camera Script' process returned errors. See plugin_octolapse.log for details". Looking into 'Debug - Camera Scripts':

`2021-03-17 19:30:29,760 - octolapse.script - DEBUG - Executing DSLR - Snapshot Camera Script with no timeout: "/home/octo/.octoprint/scripts/take-snapshot.sh" "0" "0.0" "/home/octo/.octoprint/temp-folder" "/home/octo/.octoprint/temp-folder/octolapse_snapshots_tmp/77a0b80d-03a7-44ce-8def-4f972e2d6046/354def78-9eea-409a-ad23-ee966dfff4ba" "CE3PRO_18650_batterie_holder000000.jpg" "/home/octo/.octoprint/temp-folder/octolapse_snapshots_tmp/77a0b80d-03a7-44ce-8def-4f972e2d6046/354def78-9eea-409a-ad23-ee966dfff4ba/CE3PRO_18650_batterie_holder000000.jpg" "DSLR"
2021-03-17 19:30:33,542 - octolapse.script - DEBUG - Console output (stdout) for 'DSLR - Snapshot Camera Script':New file is in location /capt0000.jpg on the camera
	Saving file as /home/octo/.octoprint/temp-folder/octolapse_snapshots_tmp/77a0b80d-03a7-44ce-8def-4f972e2d6046/354def78-9eea-409a-ad23-ee966dfff4ba/CE3PRO_18650_batterie_holder000000.jpg
	Deleting file /capt0000.jpg on the camera
	New file is in location /capt0000.nef on the camera`

This looked fine, however this strange thing popped out:

`File /home/octo/.octoprint/temp-folder/octolapse_snapshots_tmp/77a0b80d-03a7-44ce-8def-4f972e2d6046/354def78-9eea-409a-ad23-ee966dfff4ba/CE3PRO_18650_batterie_holder000000.jpg exists. Overwrite? [y|n] 
2021-03-17 19:30:33,543 - octolapse.script - ERROR - Error output (stderr) for 'DSLR - Snapshot Camera Script':
	ERROR: Could not get image.`

#### Operating System running OctoPrint and Octolapse

OS Name: OrangePi Bionic (18.04)
Os Version: Linux 4.9.170-sun50iw9

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: https://drive.google.com/file/d/1KDZRvNaNnmOK_kbATkUBuShjufGESCTg/view?usp=sharing

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://drive.google.com/file/d/1P9WHmZ5ndq8D9MJfUbSr9K2kkX3tHTYA/view?usp=sharing

#### Link to octoprint.log

Link to octoprint.log:  https://drive.google.com/file/d/1Cvm6s2ahjtwpWFL4-LWurhd9mB5wsFQ1/view?usp=sharing

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-03-17:
First, it looks like you don't have your camera set to take raw images (or maybe raw + .jpg):

> /capt0000.nef

Definitely turn that off, and only produce .jpg files.  I did add basic support for SOME raw formats, but it greatly increases rendering, capture, and transfer time, so I don't recommend it.

Second, what happens when you test take-snapshot.sh within a terminal multiple times?

WatEvsCevs, 2021-03-18:
Sorry for the late reply.
The camera was set to raw+.jpg as you suspected. When testing it within Octolapse there were no errors. 

However when running this from terminal multiple times I did get that overwrite message.

FormerLurker, 2021-03-18:
> However when running this from terminal multiple times I did get that overwrite message (but changing the filename every subsequent attempt yielded no errors).

And this happened after you turned off raw images?  The only thing I can think of is the downloaded file name isn't getting changed as requested by the script.  I'll have to think about how to handle that if this is the case.

WatEvsCevs, 2021-03-19:
> And this happened after you turned off raw images?

Yes, I changed the output mode to .jpeg. 

> The only thing I can think of is the downloaded file name isn't getting changed as requested by the script.

I'll send you the script once I get around to running OctoPrint (it's is basically a copy and paste from the external camera tutorial but something might have transferred wrong if that's the case).

FormerLurker, 2021-03-19:
> it's is basically a copy and paste from the external camera tutorial but something might have transferred wrong if that's the case

I doubt that's the issue.  Most likely your camera isn't delivering the image with the correct file name.  When you test the download, what is the file name that appears in the directory?   Also, please paste in the .sh arguments you used (I'm assuming you copy/pasted those too, but that way I don't have to look it up, lol).

WatEvsCevs, 2021-03-19:
I should clarify that I used this test command from the external camera tutorial:
`/home/pi/scripts/take-snapshot.sh 1 1 "" "/home/pi/scripts" "" "/home/pi/scripts/test.jpg"`

(which is probably why I got the overwrite error every time I ran it)

and the .sh is here:
`#!/bin/sh
# Put the arguments sent by Octolapse into variables for easy use
SNAPSHOT_NUMBER=$1
DELAY_SECONDS=$2
DATA_DIRECTORY=$3
SNAPSHOT_DIRECTORY=$4
SNAPSHOT_FILENAME=$5
SNAPSHOT_FULL_PATH=$6

# Check to see if the snapshot directory exists
if [ ! -d "${SNAPSHOT_DIRECTORY}" ];
then
  echo "Creating directory: ${SNAPSHOT_DIRECTORY}"
  mkdir -p "${SNAPSHOT_DIRECTORY}"
fi

# IMPORTANT - You must add gphoto2 to your /etc/sudoers file in order to execut$
# otherwise the following line will fail.
sudo gphoto2 --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}"

if [ ! -f "${SNAPSHOT_FULL_PATH}" ];
then
  echo "The snapshot was not found in the expected directory: '${SNAPSHOT_FULL_$
  exit 1
fi`

When doing the test print everything seems to be working fine. Looked into the snapshot tmp folder in Filezilla and saw that the files are getting renamed as required by the script. I think this issue was all boiled down to having the camera set to raw+.jpeg and forgetting to change it to .jpeg [I'm a photographer, I. like things raw :)]

FormerLurker, 2021-03-19:
Ahh, ok. Let me know if/when a real print completes for verification.  Thanks!