FormerLurker/Octolapse #714: [Request] Image Preview when Testing Camera

jbienz, 2021-03-19:
#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.3

When a user clicks the **Test Script** button in the camera profile, please consider displaying the image that was captured. This would be incredibly valuable for making sure the image is framed and focused as desired before starting the print. I know it's possible to use test mode, but that would still require waiting for each layer to take another shot. I'd really like to just click the Test button and see the results right away. It would be especially helpful if I could preview the image full-screen.

FormerLurker, 2021-03-19:
Consider enabling custom image preferences.  This has a live camera feed, which is useful.  There is even a 'Stabilize extruder' button in addition to all the image adjustment tools (perhaps focus).

I will still consider the original request, which is something I have thought about before.  I just never got around to it.

jbienz, 2021-03-19:
Yeah, I'm familiar with the custom image preferences screen and I really enjoyed using it when I was using the webcam. Unfortunately that screen isn't available for the script-based DSLR cameras.

But why would I want it? Well, many DSLRs have tiny viewfinders. Some don't support LCD at all and many that do have tiny low resolution LCDs. Mirrorless cameras generally do support HDMI out. But the Sony RX100 series (or at least the M4) refuses to do HDMI out when it's running in PC Remote mode. All the above makes it very hard to get macro things focused. Especially if your eyesight isn't perfect. 

I would love to have something like the custom image preferences screen for script-based cameras. Especially with the 'Stabilize extruder' button. Park the head where it will be, snap a frame from the camera, and see what the output will look like full screen. That would be amazing.

Thanks for your hard work and support @FormerLurker. Gonna go sponsor now.

jbienz, 2021-03-19:
P.S. If I had to choose between this one and #713 I would request this one. I can transcode videos. It's just really hard for me to get the focus right at the beginning of prints.

FormerLurker, 2021-03-19:
Ahh.. I missed the DSLR bit (didn't get much sleep, sry).  OK, it will SOMETIMES be possible to do that.  Not all scripts will result in an image being downloaded, so I need to make sure people don't always expect one to show up lest I get more error reports.  I will try to find a sneaky way to do it.

jbienz, 2021-03-19:
Totally get the sleep part! :)

And that makes total sense. Maybe set the PNotify to warning (yellow) rather than error and say something like "If your script isn't designed to capture an image, this is OK." Or something like that.

Anyway, have a great weekend!