FormerLurker/Octolapse #717: Octolapse with GRBL

Fabillous, 2021-03-20:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. I am attempting to use Octolapse in a fairly unusual way. Essentially I have a Pi camera mounted on 2 axis cnc, controlled by GRBL, for use in an image scanning setup.  Better GRBL is installed for translating the commands.

#### What should have happened?
A gcode file of a raster scan is uploaded, interspersed with G4 delays which trigger the camera.

#### What happened instead?
When Octolapse is enabled, the 'print' does not start, and eventually times out from lack of communication. It's related to the M114 command. Better GRBL translates M114 to a ?, which is the command GRBL uses for position reporting. GRBL returns the following: <Idle|MPos:0.000,0.000,0.000|Bf:15,127|FS:0,0>. Basically, is there anyway Octolapse can be configured to be compatible with GRBL?

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/Fabillous/7e4ba6cfb8b8f15132a405903ace1180

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/Fabillous/1a4e90753ec3b925b4ae114e5d885301

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-03-20:
I think I just need to accept that type of position response.  Can you send some M114 and G0 commands so I can see some various responses and learn to interpret them?  The OctoPrint terminal will do nicely for that.  I'd like to verify which axis is which (I would assume it's X, Y, then Z, but better safe than sorry).

Fabillous, 2021-03-20:
[Here's](https://gist.github.com/Fabillous/08707d71e4f267489c43d8c56dac6338) a log of connecting and then sending a few commands. Better GRBL automatically translates any M114 command sent through the terminal to the ?. It also translates any M105 command to the ?$G.

By the way, thanks so much for taking the time to have a look at this.

FormerLurker, 2021-03-25:
@Fabillous, thank you!  Hopefully I can dig into all this within the next couple weeks.  As you can tell, I have a backlog :)