FormerLurker/Octolapse #719: Octolapse intermittently taking snapshots. Gphoto2/SonyA6500

Buster-Grob, 2021-03-27:
#### If this is a feature request describe it here

Not a Feature Request

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

Started New Print using Octolapse, using GPhoto2 connected to my Sony A6500. First two snapshots work perfectly fine. Printer stabalizes, camera snaps a pic, and then the system downloads the image. 

#### What should have happened?
Ideally, it would have kept working correctly. 

#### What happened instead?
Out of 190ish planned snapshots, only about 43 were succesful, taken randomly through the print. Printer would stabalize, but the camera wouldnt take the photo. After it timed out (10s), it would show the error in the octolapse tab, and continue on. 

#### Operating System running OctoPrint and Octolapse

OS Name: Ubuntu Server
Os Version: 20.04LTS

#### Printer model & used firmware incl. version

Printer Model: Ender3
Printer Firmware Version: Unknown. Using octoprint, and the printer runs beautifully. 

#### Browser and version of browser, operating system running browser

Browser: Chrome, Octoremote, Samsung Internet
Browser OS: Windows, Android

#### Link to the gcode file you were printing when the problem occurred

 https://gist.github.com/Buster-Grob/4fbf45c11e395cc017773a572c3846a4

#### Link to settings.json

 https://gist.github.com/Buster-Grob/c8a27b70c42c59bfdfb41b487588ad4f

#### Link to plugin_octolapse.log

https://gist.github.com/Buster-Grob/6d2beef326a56ed874c2d70e4e85fba4

#### Link to octoprint.log

https://gist.github.com/Buster-Grob/eb2342cc6525a004f02cc6024ff23f81

#### Link to contents of Javascript console in the browser

Link to javascript console output: Everything in browser is working great. 

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: I dont have any. The error displayed just sayes that the snapshot failed and that I need to check the logs (or something to that effect. Sorry)

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Buster-Grob, 2021-03-27:
Something Else to note as well, now that Im trying a couple other things as well;

I tried to just have it save to the camera, and I was succesfull in getting it to snap serveral photos that way, but as soon as it runs the script you outlined, it would only take one shot every time i unplugged it from the computer. Reconnect it, and it would take one photo, either by command or by the script while printing, and then not function again. No errors, just no photos.

FormerLurker, 2021-03-28:
Ah, the dreaded sony cameras + gphoto2.  Yeah, there are lots of issues with them.  Miraculously some ppl have gotten them to work, but not without a lot of sweat and tears.  I really wish I had one of these to play with.

I recommend some google fu, looking for known issues and resolutions with gphoto2 and your camera.  Gphoto2 is continually being updated, so it's possible some of these issues are fixed on a more recent version.  You might want to try the install script and use the latest dev release of gphoto2 as well.  Can't hurt if things aren't working now anyway.