FormerLurker/Octolapse #722: replace libusb-dev by a libusb-1.0-0-dev

msmeissn, 2021-04-02:
#### If this is a feature request describe it here

Please replace libusb-dev in the gphoto install instructions to use libusb1, by specifying e.g. libusb-1.0-0-dev
(please test for the exact name.)

Reason: libusb0 is outdated, and some cameras do not work well with it, but require libusb1.

#### Version of Octolapse
 
Octolapse Version: Octolapse Wiki documentation main page

#### Version of OctoPrint

OctoPrint Version: Octolapse Wiki Documentation main page

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Getting a libgphoto2 bugreport, see https://github.com/gphoto/gphoto2/issues/409

#### What should have happened?

libusb1 should be used for building libgphoto2 instead of libusb0

#### What happened instead?

libusb0 was used, which makes e.g. Sony Alpha not work correctly.

#### Operating System running OctoPrint and Octolapse

OS Name: Linux
Os Version: 5.3

#### Printer model & used firmware incl. version

Printer Model: none
Printer Firmware Version: none

#### Browser and version of browser, operating system running browser

Browser: firefox
Browser OS: latest

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

jbienz, 2021-04-26:
@FormerLurker this issue above came from a discussion I was having with the libgphoto2 team here:

https://github.com/gphoto/libgphoto2/issues/638#issuecomment-826951577

In short, using libusb0 (which Octolapse currently does) reduces camera compatibility and also increases the amount of time it takes to acquire a frame. Getting to libusb1 would help with compatibility and could help improve print quality.

FormerLurker, 2021-04-26:
Hey all, been out of action mostly for the last few weeks, but will update the install script ASAP.  Thank you for all of your hard work and effort @msmeissn!

FormerLurker, 2021-04-27:
Ok, I've updated the script, but I have NOT notified Gonzalo Cao Cabeza de Vaca, who wrote the original script.  I'll try to do that ASAP.

jbienz, 2021-04-30:
Thank you so much @FormerLurker. Are you saying the script in the Wiki page for DSLR has been updated? And do you have any recommendations on how to upgrade an already installed Octolapse wit the new libusb1?

FormerLurker, 2021-04-30:
Yes, i updated the script.  I believe it first uninstalls all previous versions.  Don't have an alpha to test, so let me know if it works, and thanks!

jbienz, 2021-04-30:
Great! Will test next time I generate a lapse and let you know.

jbienz, 2021-05-01:
I did the update tonight. I just basically re-executed all of the steps on [this page](https://github.com/FormerLurker/Octolapse/wiki/V0.4---GPhoto2-Installation). I did backup my old install-gphoto2.sh script, and after creating the new one I did a dif. It showed the only changed line was:

`apt-get install -y build-essential libltdl-dev _libusb-dev_ libexif-dev libpopt-dev libudev-dev pkg-config git automake autoconf autopoint gettext libtool wget`

to

`apt-get install -y build-essential libltdl-dev _libusb-1-dev_ libexif-dev libpopt-dev libudev-dev pkg-config git automake autoconf autopoint gettext libtool wget`

I saw several warnings about re-linking files. I'm assuming this is because the previous files still existed, but I'm not sure. At the end it told me my version was:

`gphoto2 2.5.17`

I didn't see anything in the photo2 about or version info that would indicate which libusb was being used. I did a search and found someone recommended running:

`dpkg -l libusb-1.0*`

This returned the output:

```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name               Version      Architecture Description
+++-==================-============-============-===============================
ii  libusb-1.0-0:armhf 2:1.0.22-2   armhf        userspace USB programming libra
```

So, I think I'm good. But please let me know if there's anything else I should check and report back on.

donnysnarko, 2021-05-09:
I'm using a Sony A6500, is this fix confirmed to be working now?

FormerLurker, 2021-05-09:
@donnysnarko, I've not gotten any other verification for this fix (my camera works fine with the old libusb), but if you would like to give it a go and report back, that would be excellent!

donnysnarko, 2021-05-09:
> @donnysnarko, I've not gotten any other verification for this fix (my camera works fine with the old libusb), but if you would like to give it a go and report back, that would be excellent!

OK, i'll try giving it a try.  By the looks of things I just folllow the process [here](https://github.com/FormerLurker/Octolapse/wiki/V0.4---GPhoto2-Installation) and that should do it?

FormerLurker, 2021-05-09:
Yes, that should suffice, and thank you!

donnysnarko, 2021-05-09:
> Yes, that should suffice, and thank you!

It might be worth taking this with a grain of salt since i know nothing about Linux (but have, however, succeeded in linking a webcam for use with Octolapse), but im getting no more luck with this than before. Gphoto2 is still sending the images [to the same unknown/non-existent folder](https://imgur.com/a/aFZsaKG) .

A seperate, buy related issue is that Octolapse/Octoprint [won't detect my Sony Camera](https://imgur.com/a/7ZK4efF).  Here's the log file, not sure if it illuminates anything.
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/6448166/octoprint.log)

FormerLurker, 2021-05-09:
@donnysnarko, interesting, but that's not exactly surprising since you are using the --capture-image flag.  The process you are following contains modified scripts, so not sure what your goal is (and I don't know what the --keep flag does).

Since it looks like you are wanting to keep your images on your camera's SD card, go through [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-DSLR-With-No-Download), and be sure to follow every step.  There are probably other ways to do it, but it's easier to debug if you follow a the guide exactly.

Also, you posted octoprint.log and not plugin.octolapse.log, which isn't particularly helpful to me.  Skip that for now until you go through the no-download script and tried everything.

msmeissn, 2021-05-10:
hi, for the sub-problem ... if images are reported as being /capt<NUMBER>.suffix, it means the camera captured them only to SDRAM and libgphoto2 immediately transfered it into its virtual filesystem.
You will need to download the file now, if you cloise the libgphoto2 session it will otherwise be lost.

Note that most Sony Alpha only have this SDRAM capture mode. Only some modern ones allow selecting capture to SDCARD in the mean time.

donnysnarko, 2021-05-10:
> @donnysnarko, interesting, but that's not exactly surprising since you are using the --capture-image flag. The process you are following contains modified scripts, so not sure what your goal is (and [...]

HI, thanks for getting back with these suggestions.  I hit a hurdle at the first request of finding options for captuetarget. This is the error I'm getting
[image]
is this a Sony-related error, or something else?

msmeissn, 2021-05-16:
which libgphoto2 version is this? 2.5.24 or newer have this feature

donnysnarko, 2021-05-16:
> libgphoto2

Is there a quick command to find out which version im on?

msmeissn, 2021-05-18:
with the gphoto2 commandline tool installed:

gphoto2 --version

otherwise check with your package manager , or we also encode the version in the camlib subdirectory.