FormerLurker/Octolapse #725: Script based camera timelapse not stabilized.

kot0005, 2021-04-16:
#### If this is a feature request describe it here

no

#### Version of Octolapse
 
Octolapse Version: 0.18

#### Version of OctoPrint

OctoPrint Version: 1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. using external camera script
2.stabilization deoesnt seem to be working
3.  i have tried Snapshot delay values from 0-4000 and its all the same

#### What should have happened?
XYZ axis's should have stabilized properly at the same position each time

#### What happened instead?
stabilization occurs but its out of wack

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: [CCR10SPRO_xyzCalibration_cube.zip](https://github.com/FormerLurker/Octolapse/files/6323415/CCR10SPRO_xyzCalibration_cube.zip)

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log: [plugin_octolapse (12).zip](https://github.com/FormerLurker/Octolapse/files/6323457/plugin_octolapse.12.zip)

#### Link to contents of Javascript console in the browser

Link to javascript console output:[plugin_octolapse (12).zip](https://github.com/FormerLurker/Octolapse/files/6323457/plugin_octolapse.12.zip)

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 

Vid with poor stabilization: https://studio.youtube.com/video/57GWPmv35YI/edit
vid with good stabilization: https://studio.youtube.com/video/gUgoqxwVoso/edit

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

kot0005, 2021-04-16:
Ok this is wierd, but when i have 2 script based cameras and both runnign at the same time, everythign is fine. I will update with more findings.

I will disable 1 camera but keep them both plugged into the pi and test.

kot0005, 2021-04-17:
so i made more tests
[10:30 AM]
https://www.youtube.com/watch?v=oJb2z8vqxcY&ab_channel=Ryxxi dslr
[10:30 AM]
https://www.youtube.com/watch?v=jIhdN3u-110&ab_channel=Ryxxi 2 script cameras
[10:31 AM]
https://www.youtube.com/watch?v=TCze1Fvb7a0 just the pi camera
[10:31 AM]
am not sure why stabilization is broke when only using 1 script based camera (pi HQ)

mordhau5, 2021-04-17:
@FormerLurker Can you comment here? I've been working with them 1-on-1 to hopefully get their script-based camera working and but the script itself doesn't appear to be the issue. Triggering the script manually when the printer is still results in perfectly identical snapshots as you'd expect, but from the video of the Pi camera they linked above (https://www.youtube.com/watch?v=TCze1Fvb7a0) you can see the print head is clearly not stabilized in the same location when the snapshot is requested, it's more than just frame resonance. Their stabilization profile is also just the stock "back left", no custom stuff.  Other cameras, even other script based cameras don't seem to create this issue though.

Lot to unpack there, happy to clear things up further where I can.

FormerLurker, 2021-04-17:
My guess is that one of your two cameras is returning from the script before the picture is taken.  If you paste in the script you are using that is failing, I might be able to discover why,

FYI, the camera delay happens BEFORE the pic it taken not after.  I may need to add an after snapshot delay setting, but you can do this in your script pretty easily (just wait after the snapshot for a bit).

kot0005, 2021-04-17:
> My guess is that one of your two cameras is returning from the script before the picture is taken. If you paste in the script you are using that is failing, I might be able to discover why,
>
> FYI, [...]

hi here is my camera script in octolapse : /home/pi/scripts/PiCam-Daemon/octolapse-snap.py

i was putting together a guide on how @mordhau5 helped me set up this process but i havent finished it because of the stabilization issue.

here is the word document, some stuff might not be in order but you will prolly figure it out, let me know if u wana know anythign. i am not the expert as i just followed these steps given to me but i can try.

But my 1st script camera stabilization works normally if i have a 2nd script camera running
however it seems that second one has to be script based too, using a usb web cam breaks the stabilization again.

The first script camera here is the Pi Hiq 12.3MP (not the 2MP default res)
2nd script camera is Sony A6000

i have tried a lot of things, like different profiles, disabling all custom checks , nothing seems to work.

[Doc1.docx](https://github.com/FormerLurker/Octolapse/files/6329702/Doc1.docx)

mordhau5, 2021-04-17:
> My guess is that one of your two cameras is returning from the script before the picture is taken. If you paste in the script you are using that is failing, I might be able to discover why,
>
> FYI, [...]

I will post the script and some more details when I get to my computer later on. But for now, I will add that I am running the same exact script with a single camera and my timelapses are totally stable.

FormerLurker, 2021-04-17:
I think I see what's going on now.  The pi camera is producing stabilized results because the DSLR shot takes longer, and the hot end isn't moving until the slower DSLR shot is finished.  When you are only using the HQ camera, the script exists before the photo is acquired OR the photo that is returned is stale (only takes a few MS to throw it off).

Possible solutions: 

1. Add a millisecond delay to the END of your pi image acquisition script equal to 1/Pi_Camera_Framerate/1000;
2. Adjust the camera domain to wait 1/Pi_Camera_Framerate/1000 milliseconds before acquiring and returning an image.

In all cases your snapshot delay should be 0.

mordhau5, 2021-04-17:
> I think I see what's going on now. The pi camera is producing stabilized results because the DSLR shot takes longer, and the hot end isn't moving until the slower DSLR shot is finished. When you are [...]

That's interesting. Are you saying that octolapse is actually requesting the photo before it's ready then? Like it issues the gcode first and then immediately requests the photo before the move is done? I don't see how an image could be stale if the printer is supposed to be done moving before it triggers.

mordhau5, 2021-04-17:
If this were the case I think I could explain why my printer is stabilized and theirs isn't... My travel speed is something like 400mm/s. I bet they're using stock travel speeds for their printer profile.

FormerLurker, 2021-04-17:
> Are you saying that octolapse is actually requesting the photo before it's ready then?

Not exactly.  Just add a pause in your pi cam script i suggested above and we can verify or disprove my theory.

kot0005, 2021-04-18:
> I think I see what's going on now. The pi camera is producing stabilized results because the DSLR shot takes longer, and the hot end isn't moving until the slower DSLR shot is finished. When you are [...]

Hi! I think you are right! 

here is a test : https://youtu.be/tVynMKFbwBA

However i did not follow your suggestion, because i have no idea how to impliment them. AM waiting for  @mordhau5 to get back to me.

In the mean time i tried this:

I turned on both USB webcam and the Pi HQ cam with its scripts (this didnt work before)

Now i set my snapshot delay on the usb Webcam to 1500ms

and started a test and boom! its working as you can see i nthe video! 

[image]

[image]

kot0005, 2021-04-18:
Tried with 500ms on USB webcam , pi cam looks stabilized

https://youtu.be/QtdH67ZmDUY

will try 300ms next

kot0005, 2021-04-18:
ok 300ms is not good, but still a lot better than with no delay on 2nd camera. so I am convinced your theory is right @FormerLurker  thanks. now to just see if it works on the camera itself

kot0005, 2021-04-18:
looks like 400 is a go as well

FormerLurker, 2021-04-18:
Excellent!  I think the ultimate solution would be to update the picam script to only wait as long as necessary.  The goal would be to have the image acquired with 0ms delay added, but I know that is tricky.  I have been putting off getting my hq cam running because I've been furiously working two new projects for the last several months, but will get to this hopefully sooner rather than later.

kot0005, 2021-04-18:
> Excellent! I think the ultimate solution would be to update the picam script to only wait as long as necessary. The goal would be to have the image acquired with 0ms delay added, but I know that is [...]

Kool thanks, I will get back to you once @mordhau5 gives me the new script and I test it

One thing that would be nice is having access to the pi HQ cam's settings available in real time, right now I have to edit a Python file and restart the camera service every time , The creator said that it was a bit hard for them to Implement camera settings in any other way. 

so is it better to add this delay to my script or have the 2nd usb webcam running with the added snapshot delay ?

kot0005, 2021-04-18:
Also one of the limitation with the cutrrent setup is , I have to keep the ssh window that the camera service is running  opened. if i close it the script wont work in octolapse.

I followed this guide to send it to background with CTRL Z and then bg but as soon as i close the ssh window, the script stops workign so i am assuming its not being sent to bg properly

mordhau5, 2021-04-19:
> I think I see what's going on now. The pi camera is producing stabilized results because the DSLR shot takes longer, and the hot end isn't moving until the slower DSLR shot is finished. When you are [...]

Hey @FormerLurker I'm finally getting around to adding configurable delays to my Daemon and Client scripts. I just want to confirm that it was your first suggestion which fixed the issue for our friend here. Octolapse was resuming normal print operation before the camera frame had been loaded into the snapshot file. If you peep my client script attached, before it ends execution, it actually does check to see if the target snapshot file exists which I've confirmed does work. My guess is that picamera actually reserves and creates the file and before it's done loading the frame data in, but the client believes the file is there and hands execution back to octolapse. I'm currently having kot0005 play around in increments of 8.4ms (1/120/1000) til we lock in the minimum delay necessary. I still don't quite understand why this wasn't an issue on my Raspi 4b+ or Pi HQ cam.

[octolapse-scripts.zip](https://github.com/FormerLurker/Octolapse/files/6333089/octolapse-scripts.zip)

kot0005, 2021-04-19:
Got it to work with 0.125 sec delay in the script, i am on a CR10s pro and @mordhau5  is on ender 5 printer i think

closed 2021-04-19 by kot0005

FormerLurker, 2021-04-19:
Awesome @mordhau5!  If you aren't opposed, I'll try to get this into the next major Octolapse release.  Thank you for your work!

mordhau5, 2021-04-19:
@FormerLurker It'd be an honor. Would you like to meet some time to discuss what it might take to get it integrated? There are a few other challenges I've encountered with getting it set up that I think we could team up on.

FormerLurker, 2021-04-19:
@mordhau5, sounds like discord could work well for that.  I'm gong to be unable to dive in until around June (ArcWelder changes and need to travel and do some manual labor for the family), so let's plan to do something around then.

mordhau5, 2021-04-19:
@FormerLurker  Yep, gotta have priorities straight. Sounds good then, maybe I'll have figured out some of the other issues between now and then. Feel free to add me mordhau5#6801 to talk anytime. Or I can add you if you leave yours here.

carcamerarig, 2021-04-19:
Looking good guys, is this edging towards full res HiQ cam images? Ive only picked up on this right at the end.

mordhau5, 2021-04-20:
> @FormerLurker If possible I was suggesting mordhau5 to make a octopritn plugin that starts the camera service(python script for hi q camera), so users dont have to do it manually in a ssh window. So as long as this extension is enabled, the server will start with the pi and stay in background. Also a camera settings plugin/extension may be ? because right now it has to be done in the python script usign notepad++

Really glad it's working for you @kot0005. Ideally this wouldn't need to be a seperate plugin, and just part of Octolapse's functionality, but I don't see why it couldn't be it's own plugin. It could certainly have uses beyond Octolapse.

> 
> 
> Looking good guys, is this edging towards full res HiQ cam images? Ive only picked up on this right at the end.

@carcamerarig this is the full res image, and MUCH faster than using raspi-still.

kot0005, 2021-09-13:
[Rasp pi High Quality Camera with octolapse.docx](https://github.com/FormerLurker/Octolapse/files/7150694/Rasp.pi.High.Quality.Camera.with.octolapse.docx)

just uploading a newer version of the guide document, some people have been askign me. apparently one person got theirs to work from thsi thread too.

mordhau5, 2021-09-23:
That's crazy that other people have managed to work through all these complexities, no doubt thanks to your guide @kot0005