FormerLurker/Octolapse #726: Octolapse not 0.4.1 not extruding with prusaslicer 2.3.1

melvinmicallef, 2021-04-22:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: _1.5.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Upload gcode file sliced with prusaslicer - slicing with cura works ok, issue is only with prusaslicer
2. Print
3. Accept snapshot preview

#### What should have happened?
Extruder should extrude

#### What happened instead?
Extruder was retracting before extruding (making the extrude not push any filament)

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 1.5.3

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://pastebin.com/m157drcf

#### Link to settings.json

Link to settings.json with all passwords removed: https://pastebin.com/zLCWRaVF

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/MK1P5kG8

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/jeNDqhjV

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Joaco82, 2021-04-22:
m

melvinmicallef, 2021-04-29:
For any

closed 2021-04-29 by melvinmicallef