FormerLurker/Octolapse #727: Printer stops mid Print - Timelapse with DSLR

lenony, 2021-04-27:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.4.1

#### Version of OctoPrint

OctoPrint Version: Version 1.6.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

Was trying to print and to a timelapse with DSLR.
Printer stops printing randomly.

#### What should have happened?
Normal Printing without stops and a nice timelapse :)

#### What happened instead?
Printing stops randomly

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPI
Os Version: 1.6.0

#### Printer model & used firmware incl. version

Printer Model: Prusa Mini
Printer Firmware Version: i dont know

#### Browser and version of browser, operating system running browser

Browser: Opera
Browser OS: 75.0.3969.218

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/lenony/496b2148a9d91b12b95fa12da2037d8e

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/lenony/9988173dfc9fa3f7a28b3b07cde54453

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

lenony, 2021-04-27:
Also, i kinda found a "FIX" when the printer stops printing. Everything stays the same. RPI is still online, Interface still works.
But when i plug out the Cable of my camera the print resumes.

Does this have something to do because i use a Sony Camera ?

Camera i'm using : Sony Alpha 6000

Followed your video tutorials how to use a DSLR with Octoprint. Works fine, but somehow the printer stops randomly.

FormerLurker, 2021-05-03:
Sorry this took me a while to get to.  I tend to get to DLSR issues last, since they are so hard for me to debug.  However, this time you are in luck!  I recently received an issue from the maintainer of gphoto2 (excellent program, thanks to the author!) that addresses issues with your camera that have been plaguing users (and my github, lol) for a LONG time.  There is an update to a usb driver that apparently fixes this issue, though I don't have an Alpha to test.  I [updated the gphoto2 installation script in this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---GPhoto2-Installation) (see *Step 2 - Create the GPhoto2 Installer Script*), and running it again may solve your issue.  Please let me know!

lenony, 2021-05-06:
Hi, thanks alot. 

I will re run the Script, and we'll see if the problem still appears :)