FormerLurker/Octolapse #728: Octolapse rendering failed Rendering failed for camera 'Webcam - Default OctoPi 0.16.0'. render-error: Unknown render error.. Inner Exception: line contains NULL byte Inner Exception: line contains NULL byte

victaru, 2021-04-28:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version:  0.4.1

#### Version of OctoPrint

OctoPrint Version:  1.6.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. Printer completed printing
2. Tried to download octolapse video. Warning says "1 Unfinished rendering"
3. Clicked in, video in question status is incomplete. Used "render with selected settings"
4. Error: Rendering failed for camera 'Webcam - Default OctoPi 0.16.0'. render-error: Unknown render error. Please check plugin_octolapse.log for more details.. Inner Exception: line contains NULL byte

#### What should have happened?
Should have rendered the video for download

#### What happened instead?
Error as per #3. 

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Ender 3 Pro
Printer Firmware Version: Marlin 2.0.7.2 (running on BTT SKR Mini E3 V1.2)

#### Browser and version of browser, operating system running browser

Browser: Version 90.0.4430.85 (Official Build) (64-bit)
Browser OS: Google Chrome

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://drive.google.com/file/d/18TN0XpsxxDQrwyFY8P1335QAf8hQz8QD/view?usp=sharing

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-04-28:
I don't have access to the log file.  Please use gist.github.com to store the log and drop a link in the comments.

FormerLurker, 2021-04-28:
Also, please attach your setting.json file at gist.github.com as well.  There are instructions for reporting errors in the wiki which shows you how to get a copy of that file.

victaru, 2021-04-28:
Hi there, thanks for replying. I've updated my original post with the links to the logs on gist.github.com

staudisr, 2021-09-15:
Was this issue resolved I’m having the same issue. My octolapse use to work but had it turned off for a while and now I’m getting this issue.

FormerLurker, 2021-10-08:
Man, am I behind on issues, lol!  Sorry all.

So, I FINALLY got around to looking at the settings.json and trying to look at the rest of the attached files, but they have been encoded somehow, and are pretty much illegible!

However, I'm going to be running some tests on a new setup here soon.  What I really need to know is how the rendering profile is configured.  Could be that something changed with ffmpeg or avconv (depending on what you are using) OR maybe it's a problem with a new codec or something.