FormerLurker/Octolapse #729: DSLR working... but not working

MagmaPrints, 2021-04-28:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version:  0.4.1

#### Version of OctoPrint

OctoPrint Version:  Octoprint v. 1.6.0
                               Octopi v. 0.18.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled:  YES

#### What were you doing when the problem occurred

1.  Setup octolapse using this video  https://www.youtube.com/watch?v=J5SskYwP5JQ&ab_channel=FormerLurker
2. Everything works, test script button works without a problem and everything looks fine
3. When printing starts the dslr takes pictures, i can hear it but nothing comes on the screen and no video is rendered at the end

#### What should have happened?
DSLR should take snapshots and render into timelapse video

#### What happened instead?
Everything looks like it should be working, even the test script always gives a green result 
I can hear the picture-taking sound from the dslr when it should be taking snapshits but nothing gets saved or rendered

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi 
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Ender 3 Pro
Printer Firmware Version:  1.1.6.2

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/MagmaPrints/34630fd39774d8e8b1b0b28c048ada10

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/MagmaPrints/34630fd39774d8e8b1b0b28c048ada10

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/MagmaPrints/34630fd39774d8e8b1b0b28c048ada10

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-04-28:
Your camera is not returning .jpg images.  Check the settings and disable raw images.  Make sure ONLY jpg images are created, and it should start working.  Let me know how it goes.

MagmaPrints, 2021-05-03:
Turned off raw image. Works perfect. Thank you so much! :D

FormerLurker, 2021-05-03:
Excellent!  Closed :)

closed 2021-05-03 by FormerLurker