FormerLurker/Octolapse #730: Octoprint Crash with trigger "Smart - Snap to Print"

LuckyJaSZ, 2021-04-30:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.4.1

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.6.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Upload the GCODE
2. Octolapse setting - Trigger - Smart Snap to Print
3. Print
4. Octoprint crash
5. sudo service octoprint restart
6. 1-5 again

#### What should have happened?
Start printing

#### What happened instead?
Printing did not begin, after a while the octoprint crash - The OctoPrint server is currently not running

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPI
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: ORIGINAL PRUSA I3 MK3S & MK3S+
Printer Firmware Version: 4.3.1

#### Browser and version of browser, operating system running browser

Browser: Google Chrome - 90.0.4430.93
Browser OS: Windows 10 Pro - 19042.928

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://github.com/LuckyJaSZ/Issues/blob/main/textured-vase-by-slimprint_0.15mm_PETG_MK3S_7h1m.gcode

#### Link to settings.json

Link to settings.json with all passwords removed: https://github.com/LuckyJaSZ/Issues/blob/5d335dfd65e31a1bfc0a73144f1b73800db756b2/settings.json

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  too big file
https://github.com/LuckyJaSZ/Issues/blob/25b18247a5ca7399e0f3f894dc075891ad6b5773/plugin_octolapse.zip.001
https://github.com/LuckyJaSZ/Issues/blob/25b18247a5ca7399e0f3f894dc075891ad6b5773/plugin_octolapse.zip.002
https://github.com/LuckyJaSZ/Issues/blob/25b18247a5ca7399e0f3f894dc075891ad6b5773/plugin_octolapse.zip.003
https://github.com/LuckyJaSZ/Issues/blob/25b18247a5ca7399e0f3f894dc075891ad6b5773/plugin_octolapse.zip.004

#### Link to octoprint.log

Link to octoprint.log:  https://github.com/LuckyJaSZ/Issues/blob/a600a90c00cf1408ec6c194bec3c849a67beedfe/octoprint.log

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-05-03:
Did you perhaps upgrade to python 3?  I'll dig into the logs ASAP, but there is a LOT to go through, so it may take a while.  FYI, you can clear the octolapse log in the loggin profile.  If you have time, it would be great if you could clear it out, start a print and let it crash, then reboot and download the (much much shorter) logfile.  That way I can bypass a ton of searching.

Also, thanks for filling out a very complete issue!  I REALLY appreciate that!

LuckyJaSZ, 2021-05-03:
Hello,
I still have the same version. I have an Octoprint for a short time.
It is not a problem. So I have to:
1. Turn on the printer and allow the Octoprint and Octolapse to boot
2. Set up printing
3. Clear log
4. Turn on printing
5. After the crash and loading the Octoprint, download and save the log? Or remove the SD card after the crash and don't let the octoprint reboot? = download log to computer? What will be better?

FormerLurker, 2021-05-03:
For step 5 above:  After the crash, reboot your raspberry pi (no need to do anything with your SD card) and download the octolapse log.

Step 6:  Change your 'Logging Profile' to 'Errors Only' so you aren't logging so much.  That should save you some disk space :)

Thanks!

LuckyJaSZ, 2021-05-04:
Hello, 
here is new log
https://github.com/LuckyJaSZ/Issues/blob/main/plugin_octolapse.log
https://github.com/LuckyJaSZ/Issues/blob/main/octoprint.log

FormerLurker, 2021-05-04:
Thank you!  Will take a look hopefully this weekend

LuckyJaSZ, 2021-05-04:
> Thank you! Will take a look hopefully this weekend

I also found out that when I print a smaller object (by a lot), there is no problem. I can try reinstalling Octoprint later this weekend.
Thanks for your work :-)