FormerLurker/Octolapse #736: Octolapse with DSLR outputting in 160x120 resolution

Koyodee, 2021-05-13:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: v0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.6.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: For one print yes, for the other no. Both ran into the same issue. 

#### What were you doing when the problem occurred

1. I followed the getting started guide
2. I followed the DSLR setup guide and everything worked well
3. The camera connects and takes a picture as it should

#### What should have happened?
The rendered timelapse should be in high-quality

#### What happened instead?
The rendered time-lapse is in a low resolution of 160x120, and the preview is also blurry

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 MK3S
Printer Firmware Version: V3.10.0

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: 90.0.4430.212

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/Koyodee/38834681333bc5faba7422538010aef8

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/Koyodee/3038e0e080cb84bbbccb4ef9d5815693

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/Koyodee/aad9a8c3032ff3868c8d49577dc5c690

#### Link to octoprint.log

Link to octoprint.log: https://gist.github.com/Koyodee/a8ed4f9c254a8fe58f0d3b87e846d4d4

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://imgur.com/a/MrPHyhL

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Koyodee, 2021-05-13:
Update: Camera somehow changed back to Raw mode, changing to JPEG fixed the issue

closed 2021-05-13 by Koyodee

FormerLurker, 2021-05-13:
Hi there!  Glad you got this straightened.  DSLRs are a bit tricky, and you are not alone in having issues with the image format  :)