FormerLurker/Octolapse #739: Custom Page doesn't recognize Logitech C920 - Unknown Webcam

mapotter99, 2021-05-30:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.6.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

1. Open Camera Settings (to adjust brightness, contrast, etc)
2. Click on "Switch to custom page"

#### What should have happened?
The custom page should change to the page for the Logitech C920 camera.

#### What happened instead?
A red box appears that says "Unknown webcam
There is no custom control for your webcam model at the moment!"

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 MK3S+
Printer Firmware Version: 3.10.0

#### Browser and version of browser, operating system running browser

Browser: Firefox, Edge, Chrome
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: N/A

#### Link to settings.json

Link to settings.json with all passwords removed: https://www.dropbox.com/s/3qjsdt3gclp5jrt/settings.json?dl=0

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/mapotter99/0e9fc11a5743508bc965c5daaaaf52df

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/mapotter99/58cd880a46f2e97e25d121c755dd7fbc

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-06-01:
Thanks for separating this issue out.  Link to input.json here instead of the other issue, thanks!

mapotter99, 2021-06-01:
Here is the input.json. 

Thanks!

https://gist.github.com/mapotter99/ae56b5d5405044f9b5eee3c9bf285630

mapotter99, 2021-06-01:
Just a heads up that it is possible I have a Chinese knock-off. I got it on Amazon Prime, but Amazon wasn’t the seller. I just noticed some of the reviews said it was a knock-off. That said, Logitech G-Hub on my PC recognizes it just fine, so I don’t know. Anything is possible Give how hard webcams have been to find during the pandemic. 

Regardless, I’ve ordered another that lists Amazon as the seller. 

I just realized this, so wanted to let you know for full disclosure. If you know of a way to know for sure, let me know.

mapotter99, 2021-06-01:
So, the model of my C920 is VU0060. It looks like this model was manufactured in the Fall of 2020. The cord tag does say “Logitech Europe” on it, so regardless, I’m not sure it is a North America model, but, again, the pandemic has wreaked havoc on the webcam supply chain, so who knows. 

I’m on my way to pick up a C920S from Best Buy. 

I’ll post the input.json for the C920S in the other thread that is paralleling this one where the C920S has already been brought up.

mapotter99, 2021-06-01:
So… the VU0060 is the C920S! Odd, since the one I got on Amazon didn’t mention the “S”. 

I just picked up my new C920S at Best Buy, and it has the same model number. I may return it next time I’m near Best Buy, but I’ll hold on to it for now. 

That said, the input.json file I provided should be correct for a C920S.

FormerLurker, 2021-06-01:
You could just run dual webcams.  The possibilities are endless :)

squiddy1234, 2022-03-10:
I have the exact same problem,  "Unknown webcam,
There is no custom control for your webcam model at the moment!" (that’s just after watching Former Lurkers video on YouTube telling and showing me that there IS a custom control because I’ve seen it) 
Same camera  Logitech C920 HD Pro Webcam, Full HD 1080p/30fps Video Calling, Clear Stereo Audio.
Same seller, but Amazon DID actually sell it to me so i don't think it is a knock off.
Did we get anywhere with the original problem?
I bought the camera 8th Feb 2022 and have had nothing but trouble with it.
I’m lucky if i get 3 frames a second let alone 30.
I have followed Former Lurkers guides to the letter but NO custom control.
I have probably read everything he has written online too.
when i adjust the settings in Octolapse it takes several minutes before anything happens. Ah ha, wrong camera address? Nope.
funny though, a plugin called Camera Settings  Taylor Talkington
Interactive camera settings via v4l2-ctl, found my camera’s name and listed it correctly, before of course I disabled all other camera plugins including the spaghetti cop with the funny, revolving, Sherlock Holmes hat.
Mind you, I’m having so many problems with Octoprint, Octo pi and Octolapse it's hard to figure out what to try next to get a decent video stream. that’s all i really want, just to be able to see my nozzle on my touch screen which is connected to my PI 4B.
I know a lot of people say what’s the point in having a video screen next to your printer when you must be standing there by your printer to watch it. For me there is every point, my printer is enclosed in a coloured Perspex housing making it impossible to see if those precious first layers have stuck to the board. I also would love to make some of those beautiful time lapse thingies too.
Seriously though, if I could just get everything to boot up just for once together, I would die a happy man. I have reprinted my Elecrow 7inch screen case again, this time creating a slot for the removal of the SD card so I can get it out and re flash it with octoprint without having to unscrew everything. 
 I think I have compound problems and very little knowledge or hope of getting everything to play nicely together with my Snapmaker 2. I would be so eternally grateful for any advice that would allow me to have a day of printing and viewing instead of getting that sickening feeling every time I try to log into Octoprint or start TouchUi that I get when staring at blank screens or messages from Putty saying Octoprint  refused my connection request. How dare it !!