FormerLurker/Octolapse #740: Heatbed + extruder not returning to the same position when taking snapshot

mikserf, 2021-06-01:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

1. Started the print
2. Watched the timelapse video
3. 

#### What should have happened?
The heatbed and extruder should be at the same position on every snapshot.

#### What happened instead?
The heatbed and extruder was on different positions for every snapshot.

#### Operating System running OctoPrint and Octolapse

OS Name: Windows
Os Version: 10

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 mk3s
Printer Firmware Version: 2.3.1

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://imgur.com/SQSbHDS

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-06-01:
I need more info to solve this.  Your settings.json file would be helpful, which you can export from the Octolapse settings page (click the gear icon on the Octolapse tab, export is at the bottom).  The make and model of your camera would also be helpful.  It's probably an inadequate delay.

mikserf, 2021-06-01:
https://drive.google.com/file/d/1GVWrrPqq61nUHBpjyn9pb6Rpp1cl5uSc/view?usp=sharing

So the camera i have is Sony A7R II

FormerLurker, 2021-06-01:
It's extremely useful to know this is a DSLR.  Almost all models will not return from the 'CaptureImageAndDownload' call until the image is acquired.  It looks like the A7 returns BEFORE the image is acquired.  My guess is it's returning maybe 30-50MS too early.

The camera delay setting in the camera profile is applied BEFORE the image is taken, and is intended to account for shaking and vibration, or for streaming servers that don't always return the current frame, so that setting won't help.

However, you can augment your capture script (the .sh file) like so:

[image]

It will definitely look better, but 0.05 seconds (50 MS) may not be enough.  You want this number as low as possible without having any shaking in the final timelapse.  I recommend using test mode so you don't waste time/filament figuring out what the delay should be.  Let me know how it goes!

mikserf, 2021-06-01:
Alright, thanks!
So am I in the right place, 

General Options
Snapshot Delay - ?

Mine says 125 MS. Should I increase this? Because I did not understand if the value you highlighted was my value.

The screenshot you included in the previous answer, is that value from my settings file or ?

FormerLurker, 2021-06-01:
> General Options Snapshot Delay - ?

No, that won't work.  As I tried to explain in my first message, that delay is applied BEFORE the image is acquired.  You should set that to 0 MS most likely if you are using a DSLR.

The 'setting' I highlighted is not really a setting at all.  I added one line to the *take-snapshot.sh* file, which you should have created when following the DSLR tutorial.  It will add 50MS delay AFTER your camera is sent the 'take snapshot' command.  You may need to increase/decrease that.

In summary:  You will need to edit your *take-snapshot.sh* script (or whatever you called it) and add the *sleep 0.05* command right after the call to gphoto2 (see my snapshot).  Save the changes, and run some tests to see if the delay is high enough.

Let me know if that makes sense.

mikserf, 2021-06-03:
Will try this out this weekend. I am running the octoprint on windows, this does not change anything right?

FormerLurker, 2021-06-03:
> I am running the octoprint on windows

Yes, if by that you mean octoprint is installed on a windows machine, and not just that you are using a browser on a windows machine to access OctoPrint running somewhere else on a linux flavor.  Ug, everything is so complicated, lol!

So, assuming OctoPrint is installed on a windows machine, that would mean you can't be using gphoto2 (unless you are running octoprint in a linux VM within a windows machine, or on Windows Subsystem for Linux, LOL!!).  What software are you using to interface with your camera?