FormerLurker/Octolapse #741: [request] please add support to canvas3d.io slicer

davop, 2021-06-02:
#### If this is a feature request describe it here

Please add canvas3d.io slicer support to wonderful Octolapse
I need it to add colour to my prints but cannot then use octolapse.
That would be much appreciated

#### Version of Octolapse
 
Octolapse Version: ___0.4.1

#### Version of OctoPrint

OctoPrint Version: ___1.6.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-06-02:
> I need it to add colour to my prints but cannot then use octolapse.

I bet you can, just not with the 'automatic slicer settings' profile.  However, I'll look into it.  I have one other slicer on my list.  Can you post a small gcode file so I can take a look at how (if) it includes the slicer settings within the gcode?

davop, 2021-06-04:
Hi, Thank you, I will search for a way to use it then.

here is a link : 
https://share.canvas3d.io/projects/120ace6d8a986864e283d81aafb900c4

related code : I hope it's not too big

O21 D0014
O22 De06057972b249f09
O23 D0001
O24 D0000
O25 D1006400Dark_Green_Filament D1000000Black_Filament D1ff8c00Orange_Filament D0
O26 D004f
O27 D001e
O28 D0001
O29 D0000
O30 D0 D446db316
O30 D2 D448572f3
O30 D0 D44984f03
O30 D2 D44a71c8b
O30 D0 D44b9fdc0
O30 D2 D44c8e536
O30 D0 D44dbd083
O30 D2 D44eacefe
O30 D0 D44fdded4
O30 D2 D45067e71
O30 D0 D451037ef
O30 D2 D4517cd60
O30 D0 D4522147b
O30 D2 D4529bbf1
O30 D0 D4534aa27
O30 D2 D453c5790
O30 D0 D45474e7a
O30 D2 D454effc2
O30 D0 D45587f60
O30 D2 D45602abe
O30 D0 D4569ab56
O30 D2 D45715b7c
O30 D0 D457b2196
O30 D2 D45816813
O30 D1 D4584b998
O30 D0 D45899360
O30 D2 D458d016c
O30 D1 D4590983c
O30 D2 D459406e4
O30 D0 D4598da87
O30 D2 D459c4a54
O30 D1 D459fdddc
O30 D2 D45a34c03
O30 D0 D45a814ae
O30 D2 D45ab826f
O30 D1 D45af2465
O30 D2 D45b2923e
O30 D0 D45b74d5c
O30 D2 D45babb0a
O30 D1 D45be67fe
O30 D2 D45c1d461
O30 D0 D45c687d0
O30 D2 D45c9f268
O30 D1 D45cda204
O30 D2 D45d1097a
O30 D0 D45d5bce4
O30 D2 D45d9243a
O30 D1 D45dcd0d6
O30 D2 D45e037f4
O30 D0 D45e4d7ac
O30 D2 D45e83ec4
O30 D1 D45ebedaf
O30 D2 D45ef5194
O30 D0 D45f3f240
[... 889 lines cut ...]
; chroma inserting sparse tower layer
; transition tower auto-brim
; brim loop
; path_width=0.400000
G1 X155.4 Y133.819 F9000
G1 Z0.2 F240
G1 E1 F2100 ; destring prime
; Reset extruder pos
G92 E0
G1 X155.4 Y134.002 E0.00608 F1500
G1 X155.4 Y134.55 E0.02432 F1500
G1 X155.4 Y135.464 E0.05472 F1500
G1 X155.4 Y136.744 E0.09729 F1500
G1 X155.4 Y138.389 E0.15201 F1500
G1 X155.4 Y140.4 E0.21889 F1500
G1 X155.4 Y180.619 E1.55657 F1500
G1 X94.6 Y180.619 E3.57879 F1500
G1 X94.6 Y133.819 E5.13536 F1500
G1 X148.644 Y133.819 E6.93289 F1500
G1 X150.709 Y133.819 E7.00154 F1500
G1 X152.398 Y133.819 E7.05772 F1500
G1 X153.711 Y133.819 E7.10141 F1500
G1 X154.649 Y133.819 E7.13261 F1500
G1 X155
[Grogou-3-coul-dentifrice-by-Davop.zip](https://github.com/FormerLurker/Octolapse/files/6596719/Grogou-3-coul-dentifrice-by-Davop.zip)