FormerLurker/Octolapse #743: Octolapse on Ender 3 v2 Stringing Issue - Need Help

bobster316, 2021-06-15:
Ender 3 v2
Octolapse Version: 0.4.1
Settings Version 0.4.0
OctoPrint Version: 0.18.0
Cura 4.9.1

When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

No corresponding entry in logfile at error occurance
1.	Activating octolapse tab in Octoprint
2.	Klick on button "Renderings ..."

What should have happened?
Clean rendered 3d print while using octolapse with no stringing and blobs on the z axis.

What happened instead?
Strings and blobs caused by octolapse at the z axis where the print head moves away and back to the print when taking a photo. Same issue if changed from back left, back right, etc..
When printing object without using octolapse there is no issue with the print. This issue began approx 2 weeks ago.

Operating System running OctoPrint and Octolapse
OS Name: Octopi
Os Version: 0.18.0

Browser and version of browser, operating system running browser
Browser: Chrome
Browser OS: Windows 10

Link to contents of Javascript console in the browser
Not sure how to do this 

Screenshots and/or videos of the problem:
https://youtu.be/4PDp7ZO0iQI
https://imgur.com/HqIfcrt
https://imgur.com/U0p8Nap
https://imgur.com/73pSOFj
https://imgur.com/4tXUM9K
https://imgur.com/Pvsp8xF

bobster316, 2021-06-15:
I have files for my cura profile and octolapse settings
https://drive.google.com/file/d/1UHDWE5CZFa2fsMRKwGsssmEwKpndyPDT/view?usp=sharing
https://drive.google.com/file/d/1bMgE-q5Socx1dcOfhpPMwaWt05DPkKia/view?usp=sharing

FormerLurker, 2021-06-15:
First, have you read [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting-Print-Quality-Issues)?  If not, read through that first and pay close attention to the section on cura automatic slicer settings detection, and the g90/g91 influences extruder setting.  thanks!

bobster316, 2021-06-18:
Thanks for the advice, i went through all the documentation (hope i didnt miss anything) 
Tried with cura automatic slicer settings detection, and the g90/g91 influences extruder setting
Here a video of the result with tool temp @ 200 and  bed temp @ 70
https://www.youtube.com/watch?v=z01HrE5OmMc