FormerLurker/Octolapse #753: OctoPrint 1.6.1+Octolapse 0.4.1: midway through print, heater for hot end and bed shut off

intelligo1466, 2021-07-07:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.6.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Uploaded GCODE file to OctoPrint.
2. Started print. 
3. Accepted OctoLapse's snapshot plan.
4. Hot end and bed automatically preheated to target temperatures, as expected.
5. After reaching target temperatures, printing began normally.
6. Printing proceeds normally.
7. Some time midway through the print (20 - 40 minutes), the heaters for the hot end and bed shut off, temperatures of the hot end and bed fall, but the extruder and print head continue printing as if nothing is amiss. This happened with a variety of prints, but I haven't figured how long OctoLapse waits before shutting off the heaters.
8. I cancel the print job.
In addition, I disabled every 3rd-party plugin and the print ran normally without problems. I then enabled Octolapse as the only 3rd-party plugin and the problem described above occurred. Afterward, I disabled Octolapse and the print ran normally.

#### What should have happened?
The heaters for the hot end and bed should have continued operating according to the GCODE until the end of the print.

#### What happened instead?
The heaters for the hot end and bed shut off midway through the print job.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Monoprice MP Select Mini Pro
Printer Firmware Version: Motion Controller: v50, possibly v51

#### Browser and version of browser, operating system running browser

Browser: Google Chrome Version 91.0.4472.124 (Official Build) (64-bit)
Browser OS: Windows 10 64-bit

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/intelligo1466/d4c75e4e5cf320ff1b69f5af1e1d7377

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/intelligo1466/3d1c62e4e12abfd3d9f92777e3612bbf

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/intelligo1466/de613d0f049613ceff069932acc73c7c

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/intelligo1466/73be9012eb2709af32234a252bf33195

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-07-07:
Can you change the logging profile to 'debug' and run a test print, then post the plugin_octolapse.log file?  Not enough was logged to figure out what's going on unfortunately :(

I would suggest using test mode, but it's possible the issue won't show up since it's related to the heaters.

Thanks for posting this!

intelligo1466, 2021-07-10:
I'm afraid I must be doing something wrong. This latest plugin_octolapse.log hardly has any more information than the last one. In OctoPrint Settings, under Logging Levels I selected octoprint.plugins.octolapse and DEBUG, which resulted in the log I attached.

For this run with OctoLapse, I used the Virtual Printer plugin rather than my real printer, but I don't think it made a difference. Please advise. Thanks!

[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/6794487/plugin_octolapse.log)
[octoprint (1).log](https://github.com/FormerLurker/Octolapse/files/6794489/octoprint.1.log)

FormerLurker, 2021-07-10:
Try the logging profile on the octolapse tab (not octoprint settings).

intelligo1466, 2021-07-10:
Here's my second attempt, this time running OctoLapse with my real printer. The heaters shut down as described 36 minutes into the print. As before, the plugin_octolapse.log doesn't have much in it.

[octoprint (1).log](https://github.com/FormerLurker/Octolapse/files/6794535/octoprint.1.log)
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/6794536/plugin_octolapse.log)

intelligo1466, 2021-07-20:
Sorry it took so long to get back to you. Same error as befopre. As requested, attached are:

octoprint.log
plugin_octolapse.log
Logging_Profile.json (renamed to Logging_Profile.json.LOG for attachment) 

[octoprint.log](https://github.com/FormerLurker/Octolapse/files/6844987/octoprint.log)
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/6844988/plugin_octolapse.log)
[Logging_Profile.json.LOG](https://github.com/FormerLurker/Octolapse/files/6844989/Logging_Profile.json.LOG)