FormerLurker/Octolapse #754: Error Loading Webcam Settings - There was a problem retrieving webcam settings from the server

GesaP, 2021-07-10:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version:  1.6.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO but attached is normal log file

#### What were you doing when the problem occurred

1. I set up Octoprint with an USB Webcam (not the default raspberry cam) following this guide: https://github.com/FormerLurker/Octolapse/wiki/V0.4---Enabling-Camera-Controls
2. I try accessing the settings via clicking on the pen symbol in the Octolapse tab.

#### What should have happened?
The settings should have opened and shown the stream picture.

#### What happened instead?
At the step where I would access the settings via pressing the pen button, I get the error "Error Loading Webcam Settings - There was a problem retrieving webcam settings from the server"
[image]
[image]

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Prusa Mk2.5s
Printer Firmware Version: 3.8.1

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: 
[settings.json.zip](https://github.com/FormerLurker/Octolapse/files/6795708/settings.json.zip)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  
https://pastebin.com/tg9Ve87U

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/WgmGTZDb

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-07-12:
Try navigating to the following URLs:

```
http://IP_OF_PIE_GOES_HERE/webcam/input.json
```
and

```
http://IP_OF_PIE_GOES_HERE:8080/input.json
```

Let me know if either of those work.  If they do not, try these:

```
http://IP_OF_PIE_GOES_HERE/webcam/?action=snapshot
```
and

```
http://IP_OF_PIE_GOES_HERE:8080/?action=snapshot
```

You should see an image from your webcam from at least one of the above.  Just let me know which ones work and which do not.

Thanks!

wrcqcn, 2023-04-17:
I get the same error.
http://IP_OF_PIE_GOES_HERE:8080/?action=snapshot 
and 
http://IP_OF_PIE_GOES_HERE:8080/input.json 
work ok.