FormerLurker/Octolapse #756: Cannot rotate images acquired by gphoto2 due to root ownership?

shintaroaibara, 2021-07-23:
> 

#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.6.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

1. Setup acquiring images with a DSLR as described here: https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera
2. All goes well, images can be acquired, however I wish to acquire portrait images which require 90 degree rotation
3. Modify "Rotate, Mirror, Flip..." with "Rotate 90 Degrees" for the Camera profile
4. Start Octolapse

#### What should have happened?
JPEGs written out by gphoto2 should be rotated 90 degrees and the timelapse updated accordingly

#### What happened instead?
Since the script called for sudo gphoto2, the jpegs are owned by root, and therefore permission issues prevent octolapse from transposing the image. 

Error log says "2021-07-23 18:40:33,197 - octolapse.snapshot - ERROR - An unexpected exception occurred while transposing an image for the Canon 6D DSLR camera.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/snapshot.py", line 443, in transpose_image
    img.save(snapshot_full_path)
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/Image.py", line 2169, in save
    fp = builtins.open(filename, "w+b")
PermissionError: [Errno 13] Permission denied: '/home/pi/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp/5e6ce92f-2835-4688-b303-832e8347c7b8/1c6c514b-25ab-4ea1-aec6-48236644fc93/microneedle_mount_v2_0.2mm_PLA_MK3SMMU2S_1h16m000000.jpg'"

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 Mk3S + MMU2S
Printer Firmware Version: 3.10.0

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Chrome

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

shintaroaibara, 2021-08-17:
Well, since I couldn't work it out in a better way, this was my workaround:

Within the take_snapshot.sh I have added a new line that calls on jpegtran to rotate and overwrite the output.

sudo gphoto2 --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}"
sudo jpegtran -copy all -rotate 270 -outfile "${SNAPSHOT_FULL_PATH}" "${SNAPSHOT_FULL_PATH}"

So just like with gphoto2 I gave jpegtran sudo-ability.

FormerLurker, 2021-08-17:
Hey, thank you for the update, and sorry I didn't get around to this earlier!  I may snag your solution and add it to the DSLR guides if you don't mind!

shintaroaibara, 2021-08-17:
Yes thats fine :)  I guess it would be necessary to make a few scripts and then choose whether you want 0, 90, or 270 degree rotation and have them as presets of some description.