FormerLurker/Octolapse #760: Octolapse causing Y drift on prints.

Paper-Foxx, 2021-08-03:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 0.4.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. I slice a dual material print in Cura, and send it to Octoprint
2. Start the print, and wait for the error to happen. The layer/layers it happens on, seems to be determined by when the second extruder is in use. 
3. Octolapse will move to the snapshot location, and take a picture, but the print bed either drifts, or shifts slightly during or after the snapshot. 

#### What should have happened?
Octolapse should have moved to the snapshot location, took the snapshot while holding all motors still, and then continued printing. 

#### What happened instead?
Octolapse moved to the snapshot location, and during the 500ms delay (that I set because the print bed wasn't always in place right away with the default 125ms delay) and if the snapshot is being taken with the second extruder, the bed will do one of two things. It will either drift during the snapshot, or shift/shunt slightly upon going back to printing. I can try my best to capture proof of this happening, but for the time being, I'll include a picture of the aftermath on a priming tower. It's rather odd that it would drift in the Y direction. The extruders are separate, and IDEX machine, and the printer handles the offsets. there are not any Y offsets to speak of. 

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Tenlog TL-D3 PRO
Printer Firmware Version: Unknown, printer only displays a version number for the display, and not the firmware. 

#### Browser and version of browser, operating system running browser

Browser: Edge
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://drive.google.com/file/d/1f8KgeDF26dEQWqOzAB7sNtBAlw_Hued_/view?usp=sharing

#### Link to settings.json

Link to settings.json with all passwords removed: https://drive.google.com/file/d/1qluM9oroTnPJNpXhQKNUr9JMvH3R7zsL/view?usp=sharing

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: Not needed/Not related to issue

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: Will update post with video as soon as possible

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-08-03:
This certainly could be a bug.  I actually have never had a dual extruder machine to test this on (mine has a shared extruder).  The offsets are required so that Octolapse can put the extruder in the same position for each snapshot, otherwise it will appear to jump back and forth (or left and right) if it can be seen in the snapshot.  I may need to add a setting in case the firmware handles the offsets automatically to correctly adjust after a snapshot.

Is this only happening with the second extruder (T1)?  A video of the behavior would be extremely helpful.  I might be able to figure this out with the gcode and settings you attached, provided I understand exactly what's happening, and a video should be perfect for that.

Another thing, this might take a while to solve, and will definitely require some testing on the actual hardware.  I think it can be done via the test mode to save you some filament and time.  I'll try to cut the gcode down to a few layers for the initial tests to make things easier for both of us.

Thanks for posting this!  In all the time since this feature was added, I've actually never found someone to test the dual extruder capability, so that's great!

Paper-Foxx, 2021-08-03:
I'd be more than happy to help test for you to get this issue fixed, because I'm going to making a lot of large parts, (using a second material or filament as support material), and I'd love to be able to show off the timelapses octolapse can make. 

I would better understand you saying that octolapse needs the offsets, but there is no offset in the y direction between the two extruders, but the shifting is only happening in the Y direction. And Based off what I've watched of the printer, when it actually happens, it's always been T1 (extruder 2) that allows the Y axis to shift, and it seems like any layer that ends with T0, doesn't shift. 
I'm currently at work for another 8 hours at the time of posting, but when I do get home, I'll start up one of the prints that was giving me issues, and try to record it happening. 

I don't mind testing on the actual hardware, But as far as test mode, the printer heats quick, and I have plenty of spare filament, and it'll be nice to have something physical to look at to see the issue, (as well as the fact that I don't have a particularly high quality camera, but if you're okay with grainy 1080p, I can do that). As far as cutting down the Gcode goes, it might be better to leave it larger, as if it's only happening with one extruder, it'll have to end the layer on that extruder for it to happen.

I should also mention, if you're looking through the gcode I sent, you won't find Cura's notation (TYPE;SUPPORT) Marker, because Cura wasn't setting the active extruder to T1 for supports, it was only doing it every other layer, and trying to print with T0, which was at the standby temp. As a bandaid fix for that, to force supports to print with T1 always, (which I wanted from the start), I've replaced TYPE:SUPPORT with T1 in the gcode. I thought I'd mention this, just in case this is somehow messing with Octolapse, even if it does print just fine with the plugin disabled. 

TL:DR, as soon as I'm home, I'll try and get a video of the issue happening. I'm more than happy to test and fix this.

Paper-Foxx, 2021-08-05:
I managed to get it to happen on a smaller print, and while it did happen, I had my camera poorly set up and it didn't catch anything, as well as me forgetting to have logging on in octolapse. For the time being, here is a picture of that print, and the code for it. I printed it a second time with octolapse off, and it printed perfect. 
[image]
made with this Gcode
https://drive.google.com/file/d/1cRqYcDwKzyeW88kAIscWnd8Tnnuj06Pq/view?usp=sharing
A slight observation that I did make with this failed print though, is that every time a snapshot is taken with extruder two, or T1, that the bed is slightly forwards farther by a consitent amount. That can be seen here in the timelapse made by octolapse during that print. Would that be a setting somewhere in octolapse causing this? if so, it may be the issue, as that seems to be the same amount the print is shifting by, but the confusing part is that it only happens sometimes, not all the time. Here's that clip. 
https://drive.google.com/file/d/1AsWy7m4cQzqnI9jVs8rcz-cTgkLlbda5/view?usp=sharing

It seems the issue isn't as prevalent in smaller prints, as it only happend very small, and only 3 or 4 times. I'm currently printing a larger print, which is already showing slight shifts, and my camera is set up much better this time, and I'll be able to give you a full recording of the printer and printing process. I'll send over the gcode used as well after the print. I've also turned on full logging for this print, so that you'll have as much as possible to work with. It should be done by the end of the day, or even earlier if the print fails too badly, but I've positioned it so that it can fail and shift multiple times. So I'll have a very good example of it, and video for you to review. As well as the logs.

FormerLurker, 2021-08-05:
Excellent.  I will wait for the logs.  Can you also include a close-up pic of your nozzles and measure the offset.  I will explain why this is important when I get back in town on Monday.

Paper-Foxx, 2021-11-04:
It appears I never sent my update. I apologize, as I thought I had. I recorded a video, but I can't find it at this time. I'll recreate and record the issue for you to see properly, and included the printed gcode.