FormerLurker/Octolapse #764: Gphoto2 no detect mi model camera

Valencanoextinto, 2021-08-16:
Hello is my first time that i write here, sorry if a don´t use this

I have a problem when i use "gphoto2 --auto-detect", it can't show me my model camera
[image]

I am using a canon 250D, it is in the list compatible cameras

thans so much for the help

FormerLurker, 2021-08-16:
I'm not really a gphoto2 expert, but I'll do what I can.  Is it not taking a snapshot when you send the appropriate command, or is it just not showing the model with the --auto-detect parameter?

Valencanoextinto, 2021-08-16:
Thank you very much in advance, it is the first time that I use so much code

this is what appears to me after using the command: "sudo gphoto2 --auto-detect --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}""
[image]

With him in principle I do both, right?

FormerLurker, 2021-08-16:
Your camert probably needs to be in M mode.  Search for bulb release on this page, and read about the canon eos cameras: http://www.gphoto.org/doc/remote/

You may need to add a few config lines to the script.  Octolapse can run a config script on print start, which will reduce capture time a tiny bit :)

Valencanoextinto, 2021-08-16:
Ok I have changed to manual mode and it already captures images:
[image]

but I don't understand what code of the script you mean
I'm very sorry, I don't understand much about programming :(

FormerLurker, 2021-08-16:
> Ok I have changed to manual mode and it already captures images:

Excellent!  I recommend running a test print to make sure your camera work for repeated photos.  If you get any timeout errors (bulb errors), there may be more to do.

> I'm very sorry, I don't understand much about programming :(

That's OK!  If it's taking pictures, and doesn't error out during a long print, you should be just fine as you are!

Valencanoextinto, 2021-08-16:
Thank you very much it seems that in the configuration part I have been able to do everything

in a while I will do the test mode thank you very much again as soon as I have news I will inform you

Valencanoextinto, 2021-08-17:
perfect everything works very well

thank you very much again

jonas060831, 2021-11-29:
i have an old nikon d3100 and would like to use it on 3d printer but gphoto2 does not seem to detect it when running 

``` sudo gphoto2 --auto-detect ```

is Nikon d3100 still supported on the latest version if not can i use older version when i try 

``` apt-cache policy gphoto2 ``` 

under the version table it only show 1 version available (2.5.20)