FormerLurker/Octolapse #767: [Request] Option to disable all checks for the printer

Macj332, 2021-09-29:
#### If this is a feature request describe it here

I have a 3D printer that I am using to test software on a tablet by attaching a stylus to the head of the printer. I need to be able to take snapshots of the display after certain button presses. Currently, I am unable to do this because Octolapse will not trigger from a Gcode trigger if no layer has been placed. I also needed to fill in a bunch of random data in the slicer settings area because I am not using a slicer.

It would be very convenient if Octolapse had a setting where it do nothing except take a picture when it sees a Gcode trigger.
Please let me know if this is already a setting that I just haven't found yet. Thank you.

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.6.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-10-08:
No, there is no convenient way to do this currently.  However, it would be pretty easy to setup a script and use the GcodeSystemCommand plugin to capture an image from your camera's stream.  You can even then use Octolapse to assemble the snapshots into a video by uploading them into the 'Videos and Images'->'Saved Snapshots'->'Import Snapshot Archive (zip)' functionality!  Let me know if you need some help.  The first timelapse I did was using the GcodeSystemCommands plugin, so I have a soft spot in my heart for that plugin :)

FormerLurker, 2021-10-11:
I basically downloaded the image from the snapshot url (http://ip_of_pi/webcam?action=snapshot) using curl.  It will be a pretty short and easy script.  I tried to find the one I used, but I might not have saved it :(