FormerLurker/Octolapse #770: Extruder inverts motion after 2nd layer

Apexpredation, 2021-10-10:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: v0.4.1

#### Version of OctoPrint

OctoPrint Version: v1.6.1 

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Printing a model via octoprint with Smart -high quality and animated - printing on in Octolapse
2. standard printing no special modes activated (not vase mode)

#### What should have happened?
finished print with cool timelapse

#### What happened instead?
after the second layer finished the extruder motion was put into reverse. retractions fed the filament in and printing motions retracted the filament.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: BIQU B1
Printer Firmware Version: Marlin 2.0.9.1

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://1drv.ms/u/s!AhVBZLU4JSkbhsty0Vu5CYqVfrXTOA?e=9POhvQ

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://1drv.ms/u/s!AhVBZLU4JSkbhstzX6ExLsHQgO-kWQ?e=KFPPMb

#### Link to octoprint.log

Link to octoprint.log: https://1drv.ms/u/s!AhVBZLU4JSkbhst5guCslFdVLURRuA?e=KQCkdS

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/7317444/octoprint.log)

FormerLurker, 2021-10-10:
Try checking the G90/G91 influences extruder setting in your printer profile settings.  Marlin 2 and smoothieware require this, as does prusa buddy firmware, and several others.  Let me know if that helps.

Apexpredation, 2021-10-21:
It seems to be working now. Thanks. Now I'm having another issue, my rendered .MP4 videos are unplayable when using my DSLR. All the photos are taken and I can use an online .mp4 to .gif converter and that works fine, but the original video gets error "unable to play. Unsupported file type" Any tips on that? Should I make a new bug report?

AxePlex, 2023-03-01:
Hi, i have the same issue! @Apexpredation on what settings did you put the G90/G91 its just to gain some time and don't mess it up ?? thx for your future reply

Pneumanifest, 2023-03-01:
There's a check box for "G90/G91 influences extruder" in the printer profile settings in Octoprint

AxePlex, 2023-03-01:
> There's a check box for "G90/G91 influences extruder" in the printer profile settings in Octoprint

I saw it! it's just on what should I put it? Default Absolute, Relative or try the Forced one?? 

Sorry I'm new to Octoprint and Octolapse

Pneumanifest, 2023-03-01:
It's been so long since I've used that. I don't think there was an option for others when I did it. I would stick with default.

AxePlex, 2023-03-01:
> It's been so long since I've used that. I don't think there was an option for others when I did it. I would stick with default.

ok I'll try between Default Absolute, Default Relative, will! 

I'll post if it'll work for me or not in case someone (will) have the same issue

thx for your reply

AxePlex, 2023-03-01:
i have an other question! is there a way to use the DSLR as the webcam vue when not used in timelapse mode??

Pneumanifest, 2023-03-01:
Default to absolute, let the slicer set relative if it needs it. 

My DSLR doesn't allow for live view so I can't say if that's available or not with others.

AxePlex, 2023-03-01:
> Default to absolute, let the slicer set relative if it needs it.
> 
> My DSLR doesn't allow for live view so I can't say if that's available or not with others.

ok I'll try default to Absolute! ill continue searching for the webcam mode on a DSLR

Pneumanifest, 2023-03-01:
Just look for the ability of your specific camera model. Not all have a webcam option. Mine will only show the view when I use the Canon photography program but it will not do anything with any other webcam program.

AxePlex, 2023-03-01:
> Just look for the ability of your specific camera model. Not all have a webcam option. Mine will only show the view when I use the Canon photography program but it will not do anything with any other webcam program.

where can you find this? because in all my research I never saw this!

I'm Using a Canon 200D

Pneumanifest, 2023-03-01:
Look up the EOS webcam utility or the Canon site for your model.

AxePlex, 2023-03-01:
> Look up the EOS webcam utility or the Canon site for your model.

yep but on the Raspberry its possible??

Pneumanifest, 2023-03-01:
Like I said, I don't know since mine definitely doesn't work that way

AxePlex, 2023-03-01:
ok thx for all the informations