FormerLurker/Octolapse #780: Octolapse and Lumix over Wifi

DenisDee, 2021-11-07:
#### Version of Octolapse
 0.4.1

#### Version of OctoPrint
1.7.2

So I tried to control my Lumix G80 over the wifi following the tutorial from one german youtuber. He used next:

#### Before Print Start Script
#!/bin/sh
curl -X Get -G http://192.168.54.1//cam.cgi?mode=setsetting \
-d type=device_name \
-d value=SM-G903F \

sleep 1

curl -X Get -G http://192.168.54.1//cam.cgi?mode=accctrl \
-d type=req_acc \
-d value=4D454930-0100-1000-8001-024500021C98 \
-d value2=SM-G903 \

sleep 1

curl -X Get -G http://192.168.54.1//cam.cgi?mode=setsetting \
-d type=device_name \
-d value=SM-G903F \

#### Before Snapshot Script
#!/bin/sh
curl -X Get -G http://192.168.54.1/cam.cgi?mode=camcmd \
-d value=capture  \

All tested in terminal all works. It is connected to camera, it can take snapshot. Then I move to octolapse, I write path to scripts and press test. On testing Before print starts script I get this error: Camera Script Failed
Errors were detected - The 'DSLR - Before Print Camera Script' process returned errors. See plugin_octolapse.log for details

Continuing to test Before snapshot script, when I press it camera takes picture, but I get next error: 
Camera Script Failed
Errors were detected - The 'DSLR - Before Snapshot Camera Script' process returned errors. See plugin_octolapse.log for details

And here is the plugin_octolapse.log:

> 2021-11-07 15:45:10,157 - octolapse.settings - INFO - Saving settings to: /home/pi/.octoprint/data/octolapse/settings.json.
2021-11-07 15:45:10,220 - octolapse.settings - INFO - Settings saved.
2021-11-07 15:45:14,580 - octolapse.__init__ - INFO - Attempting to update all current profile from the server
2021-11-07 15:45:14,787 - octolapse.__init__ - INFO - No profile updates are available.
2021-11-07 15:45:20,584 - octolapse.__init__ - INFO - Attempting to update all current profile from the server
2021-11-07 15:45:20,769 - octolapse.__init__ - INFO - No profile updates are available.
2021-11-07 15:45:35,737 - octolapse.__init__ - INFO - Attempting to update all current profile from the server
2021-11-07 15:45:35,928 - octolapse.__init__ - INFO - No profile updates are available.
2021-11-07 15:45:40,191 - octolapse.script - DEBUG - Executing DSLR - Before Print Camera Script: "/home/pi/scripts/lumixan.sh" "DSLR"
2021-11-07 15:45:42,422 - octolapse.script - DEBUG - Console output (stdout) for 'DSLR - Before Print Camera Script':<?xml version="1.0" encoding="UTF-8"?>
	<camrply><result>ok</result></camrply>ok,G80-28C8CB,remote,encrypted<?xml version="1.0" encoding="UTF-8"?>
	<camrply><result>ok</result></camrply>
2021-11-07 15:45:42,424 - octolapse.script - ERROR - Error output (stderr) for 'DSLR - Before Print Camera Script':
	  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
	                                 Dload  Upload   Total   Spent    Left  Speed
	
>  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100    78    0    78    0     0   8666      0 --:--:-- --:--:-- --:--:--  8666
	  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
	                                 Dload  Upload   Total   Spent    Left  Speed
	
>  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100    30    0    30    0     0   2000      0 --:--:-- --:--:-- --:--:--  2000
	  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
	                                 Dload  Upload   Total   Spent    Left  Speed
	
 > 0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100    78    0    78    0     0  15600      0 --:--:-- --:--:-- --:--:-- 15600
2021-11-07 15:45:50,406 - octolapse.script - DEBUG - Executing DSLR - Before Snapshot Camera Script: "/home/pi/scripts/lumixfoto.sh" "0" "125" "/tmp/tmpld7zspzz" "/tmp/tmpld7zspzz/42737452-d8a0-4750-a7e1-63c3e325c9ff/810ffa52-f9f0-4799-af49-ddada266b659" "test_snapshot000000.jpg" "/tmp/tmpld7zspzz/42737452-d8a0-4750-a7e1-63c3e325c9ff/810ffa52-f9f0-4799-af49-ddada266b659/test_snapshot000000.jpg" "DSLR"
2021-11-07 15:45:50,492 - octolapse.script - DEBUG - Console output (stdout) for 'DSLR - Before Snapshot Camera Script':<?xml version="1.0" encoding="UTF-8"?>
	<camrply><result>ok</result></camrply>
2021-11-07 15:45:50,493 - octolapse.script - ERROR - Error output (stderr) for 'DSLR - Before Snapshot Camera Script':
	  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
	                                 Dload  Upload   Total   Spent    Left  Speed
	
>  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100    78    0    78    0     0   7800      0 --:--:-- --:--:-- --:--:--  7800
2021-11-07 15:45:57,826 - octolapse.__init__ - INFO - Attempting to update all current profile from the server
2021-11-07 15:45:58,010 - octolapse.__init__ - INFO - No profile updates are available.

Any advice  what I can do is more then helpful.

heregoesnothing3d, 2021-12-29:
Same issue here

FormerLurker, 2021-12-30:
Ok, so curl is outputting progress info over stdError, which SHOULD indicate failure.  I'd hate to break the error detection due to a non-standard response from this, so perhaps you can modify your call like so:

```curl -X Get -G {URL_GOES_HERE} --fail --silent --show-error````

Let me know if that helps.

Also, sorry it took me SO DAMN LONG to respond..  I always get crushed with end-of-the-year projects.  Planning to take time off soon to get to some of the issue backlog.