FormerLurker/Octolapse #783: Error in runing test for Octolapse for my nikkon 750 camera in the test of script.

monojuli, 2021-11-11:
#### Version of OctoPrint

OctoPrint Version: 0.18.0

#### When you ran into the problem, did you have diagnostic logging enabled?

2021-11-11 03:19:15,037 - octolapse.script - ERROR - Error output (stderr) for 'New Camera - Snapshot Camera Script':
	sudo: no tty present and no askpass program specified
	The snapshot was not found in the expected directory: '/tmp/tmp9ciqq179/729172b4-ee5c-4d5c-ba2c-793696fa2a50/dcda9520-2eba-4e8a-aa1d-f53a11a51f53/test_snapshot000000.jpg'.

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

was running the test snapshot Acquire Script. in the botton test script
[image]

this is my script
[image]
 i have sudo visudo allright
[image]
this is whereis 
[image]

#### What should have happened?
well it should say test ok.. but puts a red flag
[image]

#### What happened instead?
up there

#### Operating System running OctoPrint and Octolapse

#### Printer model & used firmware incl. version

Printer Model:MK3S

#### Link to the gcode file you were printing when the problem occurred

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log
https://gist.github.com/monojuli/5883ae3f7c4c29f74fa50978169d4543

#### Link to contents of Javascript console in the browser

forndog, 2021-11-14:
I'm having the exact same issue

Xaldew, 2022-03-22:
Hello,

I had a quick look at the log-file, it appears to be a similar problem reported in a few other tickets: Your camera is set to download both a JPEG and a RAW file, but fails when downloading the second, since it hangs waiting for user input to allow it to erase the first one.

First, try to change your `gphoto2` script to do this instead:

```
cwd=$(pwd)
cd $SNAPSHOT_DIRECTORY
gphoto2 --auto-detect --capture-image-and-download --force-overwrite --filename "${SNAPSHOT_FULL_PATH}" 
cd $cwd 
```

Note that with this you don't need to use `sudo` anymore, and the `--force-overwrite` removes the need for user intervention as noted above.

Secondly, make sure that your cameras (*on the camera itself*) are set to only download a JPEG for now, at least until you get it to work.