FormerLurker/Octolapse #784: octoprint.plugin - ERROR - Error while calling plugin octolapse

Elimap1, 2021-11-12:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse 
 
Octolapse Version: Octolapse 0.4.1

#### Version of OctoPrint 

OctoPrint Version: OctoPrint 1.7.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. installing python 3.10.0
2. installing OctoPrint 1.7.2
3. install OctoLapse 0.4.1 Plugin

#### What should have happened?
i should start setup the Octolapse plugin for first use

#### What happened instead?
getting error message on  octoprint server as listed on title
and getting error on browser "Seting Load Error  status: error. Error. INTERNAL SERVER ERROR"

#### Operating System running OctoPrint and Octolapse

OS Name: windows 
Os Version: 10 64bit

#### Printer model & used firmware incl. version

Printer Model: Ender 3
Printer Firmware Version: marlin 2.0

#### Browser and version of browser, operating system running browser

Browser: Chrome /Edge
Browser OS: windows 10 browser

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log: (https://github.com/Elimap1/Elimap1/blob/main/plugin_octolapse.log)
https://github.com/Elimap1/Elimap1/blob/main/octoprint.log

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

cp2004, 2021-11-16:
I've seen this one around the forums once or twice now. I think it is probably the same as #772.

This is the error: `AttributeError: module 'collections' has no attribute 'Iterable'`. Until Python 3.10, these were available under both `collections.Iterable` and `collections.abc.Iterable`, but the old reference was removed.

You'll need some compatibility code to handle the Python 2 case (old style, collections.Iterable) and the new style (Python 3, collections.abc.Iterable). An example of this would be below:

https://github.com/OctoPrint/OctoPrint/blob/945825f03fbd2417a009dd33daf9f787aff60211/src/octoprint/util/__init__.py#L29-L33

Hope that helps @FormerLurker 🙂

Elimap1, 2021-11-19:
FIX
the problem is related to the python new version 3.10.0
uninstalling the v 3.10.0 and install the v 3.9 fix the problem!

FormerLurker, 2021-11-23:
I just got another report of this.  I really wish they would stop making breaking changes in minor releases... sigh..

Anyway, thanks for the report!

cp2004, 2021-11-23:
I believe it was changed for Python 3.3, but to make it more backwards compatible friendly they added a compatibility layer which just pushed the breaking change out further... It was even delayed from 3.8, to 3.9 and then finally to 3.10, but that doesn't make it any easier!

FormerLurker, 2021-11-23:
@cp2004, I just wish they'd stick to the standard version number system:  Major.Minor.Bugfix.  With python it's more like Major.SometimesMajor.Minor.

Anyway, I updated the code and have pushed to devel.  Here is a link you can use to install via the plugin manager for testing:

```https://github.com/FormerLurker/Octolapse/archive/d33011eda4bf1dead224d6a897927eaba117be2c.zip```

There are a few new features in there too:  Superslicer comment support for better quality.  Fixed un-minified some JS to support OctoPrint's built in minification.  Added customizable timestamp/elapsed time strings to support whatever format you like (rendering profile/overlay, I added help links).

Warning:  I have NOT actually tested in python 3.10 yet, since this was a quick fix.  Could be more problems.  I'll try to get to that ASAP.

FormerLurker, 2021-11-23:
All, I'm going to close this issue out since there is a better report [here](https://github.com/FormerLurker/Octolapse/issues/787) in issue #787.  Let's continue the discussion there please!  I'll call out your usernames over there if I can figure out how to do that cross issue :)

closed 2021-11-23 by FormerLurker

mjbstldc, 2022-12-12:
I had success with your fix.  Thanks!