FormerLurker/Octolapse #787: Octolapse not working with Python 3.10

insomaniac49, 2021-11-23:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.7.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

1. I was having issues with my OctoPrint so I started with a fresh install. Uninstalled Python 3.9.4, installed Python 3.10.0.
2. From here I rebuilt OctoPrint, everything is working as it should, just going through and adding my plugins.
3. All plugins installed fine except for Octolapse, the install completed and instead of the correct name in the install complete popup had "Unknown"
4. After restarting OctoPrint it started throwing an Internal Server Error
6. Attempted to reset setting to default for Octolapse, same issue.
7. Cant edit settings or do anything with the plugin currently.

#### What should have happened?
It should not have had any error popups

#### What happened instead?
Lots of error popups
[image]

#### Operating System running OctoPrint and Octolapse

OS Name: Windows 10
Os Version: 21H1

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: FireFox/Chrome
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/insomaniac49/2cce03668c7582451edf9a9bd17de94c

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  (https://gist.github.com/insomaniac49/c97f4aee1e31a42634cb37098626e544)

#### Link to octoprint.log

Link to octoprint.log: https://gist.github.com/insomaniac49/94041c68cb1765c67792e5f740a77594

#### Link to contents of Javascript console in the browser

Link to javascript console output: (https://gist.github.com/insomaniac49/c1dc1cba19531d438732a7546c13c960)

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: [image]
https://community.octoprint.org/t/settings-load-error/38005, found this while searching in regards to the error.

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2021-11-23:
Sigh..  Why do they keep breaking things in minor releases!  The Python team seems to not really care all that much about backwards compatibility, unfortunately.

insomaniac49, 2021-11-23:
Yes I understand.... anytime I upgrade/update I begin to immediately flop sweat waiting for something to break.

FormerLurker, 2021-11-23:
This is copy/pasted from the other, less detailed issue report.  I'm closing that one down and am going to invite everyone here.  

I updated the code and have pushed to devel. Here is a link you can use to install via the plugin manager for testing:

https://github.com/FormerLurker/Octolapse/archive/d33011eda4bf1dead224d6a897927eaba117be2c.zip

There are a few new features in there too: Superslicer comment support for better quality. Fixed un-minified some JS to support OctoPrint's built in minification. Added customizable timestamp/elapsed time strings to support whatever format you like (rendering profile/overlay, I added help links).

Warning: I have NOT actually tested in python 3.10 yet, since this was a quick fix. Could be more problems. I'll try to get to that ASAP.

insomaniac49, 2021-11-23:
So far so good! Able to install via plugin manager and edit settings in Octolapse again with no errors, I will have to run some test prints and verify it's working as it should.

FormerLurker, 2021-11-23:
You can use teat mode to try a dry run (no extrusion, heating or fans) for quick and easy testing.  A live print would also be nice.

insomaniac49, 2021-11-24:
I did forget about that, but I had some prints to do anyway. Both tests(1 on each printer) successfully generated an Octolapse as expected. As far as I can tell it is working like it was previously for me on Python 3.9.4. Again didn't test any new features just your basic Octolapse.

FormerLurker, 2021-11-24:
Fantastic!  I tested on 3.7, so I am going to call this one closed!  Will push to master ASAP

GossamerSolid, 2022-01-06:
> Fantastic! I tested on 3.7, so I am going to call this one closed! Will push to master ASAP

I tried this plugin link you provided because I'm having the same issue OP reported.

Upon trying to configure a new printer (I select my make and model - AnyCubic Mega-S), I get this error in the client JS console:

`packed_plugins.js?be839fff:30689 Uncaught TypeError: this.charCodeAt is not a function.  Exception occurred when checking element octolapse_printer_snapshot_command, check the 'octolapsePrinterSnapshotCommand' method.
    at PrettyGCodeViewModel.String.hashCode (packed_plugins.js?be839fff:30689)
    at i (packed_libs.js?1fe9f29c:13)
    at Dt (packed_libs.js?1fe9f29c:13)
    at Function.S.param (packed_libs.js?1fe9f29c:13)
    at $.validator.remote (packed_plugins.js?be839fff:16327)
    at $.validator.<anonymous> (packed_plugins.js?be839fff:17163)
    at $.validator.check (packed_plugins.js?be839fff:15550)
    at $.validator.checkForm (packed_plugins.js?be839fff:15227)
    at $.validator.form (packed_plugins.js?be839fff:15214)
    at packed_plugins.js?be839fff:19267`

Prevents me from saving the printer profile, which prevents me from using octolapse.

FormerLurker, 2022-01-07:
This is def an issue with the jquery validator.  I'll take a look ASAP.

GossamerSolid, 2022-01-07:
> This is def an issue with the jquery validator. I'll take a look ASAP.

Awesome, much appreciated!

FormerLurker, 2022-01-07:
@GossamerSolid, can you replicate the error, then clear the javascript console and re-replicate the error?  The only place I can find the charCodeAt function is in Showdown.js (a markdown rendering library, which is super awesome by the way), which wouldn't be related to this issue, I don't think.  However, I notice that the error was actually thrown by PrettyGCode.  Can you also try disabling that plugin temporarily to see if it starts working again?  This would go a LONG way towards narrowing down exactly what's happening.  I appreciate it!

GossamerSolid, 2022-01-07:
> @GossamerSolid, can you replicate the error, then clear the javascript console and re-replicate the error? The only place I can find the charCodeAt function is in Showdown.js (a markdown rendering [...]

I have 12 hours left on a print, so I can't right now, but I will when possible.

GossamerSolid, 2022-01-09:
> @GossamerSolid, can you replicate the error, then clear the javascript console and re-replicate the error? The only place I can find the charCodeAt function is in Showdown.js (a markdown rendering [...]

Ok, so here's what I did:

1. Re-enable plugin
2. Restart OctoPrint
3. Go to OctoPrint Settings
4. Go to Octolapse Plugin
5. Switch to "Printer" tab
6. Click "Add Profile"
7. In the "Make" dropdown, select "Anycubic"
8. In the "Model" dropdown, select "Mega-S"
9. In the notification that appears in the top right asking to update profile from server, I click "Ok"
10. Click the "Save" button
11. Doesn't save, JS console log is identical to the one I posted before.

For some information, I'm running on Windows 10 Pro x64 and have Python 3.10.1 installed.

GossamerSolid, 2022-01-12:
@FormerLurker did you have a chance to look into it. Just wondering if I should just try to downgrade my python or not.

FormerLurker, 2022-01-12:
Not yet, but it is on my agenda.  Sorry for all the delays, but I am in the middle of a huge project for a client.  Once the dust has settled, this will be my top priority, as I expect more and more users to run into it as things get upgraded.

insanemetalman, 2022-01-15:
I am having the same error that this thread started with, I had octolapse previously installed and I could pull up the UI, but I didn't have the dev tools installed, I deleted octolapes and redownloaded it, and now I get the settings load error from above.  I am running Octoprint through a windows laptop turned server for the sake of saving college money.

derekakelly, 2022-05-18:
I also just encountered this error, and after installing the updated file, it seems like everything is back to normal.

MikeOxlong83, 2022-07-09:
What were you doing when the problem occurred:
1. Upgraded python to Python 3.10.0 in the course of updating another plugin
2. All other plugins install fine except for Octolapse, the install completes and instead of the correct name in the install complete popup, it's listed as "Unknown"
3. After restarting octoprint, there is no octolapse tab, nor settings in the settings dialogue, nor entry in plugin manager.
4. Attempting to reinstall octolapse shows that it actually uninstalls the non-existant octolapse before reinstalling it and repeating the problem

Operating System running OctoPrint and Octolapse
OS Name: Mac OSX
Os Version: 10.11.6

Printer model & used firmware incl. version
Printer Model: AnyCubic i3 Mega S
Printer Firmware Version: Marlin 1.1.9

Browser and version of browser, operating system running browser
Browser: FireFox/Chrome
Browser OS: Mac OSX 10.11.6

MikeOxlong83, 2022-07-09:
[octoprint-systeminfo-20220708200359.zip](https://github.com/FormerLurker/Octolapse/files/9075764/octoprint-systeminfo-20220708200359.zip)

my octoprint bundle

jneilliii, 2022-07-29:
> This is copy/pasted from the other, less detailed issue report. I'm closing that one down and am going to invite everyone here.
>
> I updated the code and have pushed to devel. Here is a link you can [...]

@FormerLurker any chance we can get this version released official? We keep getting a fair amount of support overhead because of this 3.10 conflict.

total1ty, 2022-11-25:
Bump, got the same problem, Octolapse doesnt work anymore after upgrade to Python 3.8 or 3.10

aidan-mundy, 2022-12-10:
Also experiencing this problem on 3.10. Would love an official release, but I can confirm the zip file release also works.

mjbstldc, 2022-12-12:
I had success with the fix by @FormerLurker Thanks!!

SteelTitan1010, 2022-12-15:
Is there installation introductions anywhere, I have been trying but haven't had any success yet?

aidan-mundy, 2022-12-15:
@SteelTitan1010 Take the .zip link from above and paste it into the "install from link" field of the plugin manager. I'd also recommend uninstalling the broken Octolapse version first.

jneilliii, 2022-12-28:
Would love to get these linked changes merged and released official @FormerLurker, python 3.10 compatibility is causing a good amount of support overhead with more and more people installing in windows and other OS, etc.

Nathan22211, 2023-02-04:
Having some issues myself with the plugin on python 3.10. The add profile button doesn't work at all. I have it hooked up to a zonestar but I think this is sided on the server, so that shouldn't matter

2probro, 2023-02-20:
This is still an issue, I cannot use octolapse anymore. Please fix this

2probro, 2023-02-20:
I fixed the problem by uninstalling octolapse and reinstalling it via the link provided by @FormerLurker 

> https://github.com/FormerLurker/Octolapse/archive/d33011eda4bf1dead224d6a897927eaba117be2c.zip

[image]

viorellu, 2023-03-02:
I can confirm using the version from the link provided by @FormerLurker works on Ubuntu on a x86 PC

https://github.com/FormerLurker/Octolapse/archive/d33011eda4bf1dead224d6a897927eaba117be2c.zip