FormerLurker/Octolapse #791: Trying to install on Octo4a and crash

nandobarreto, 2021-11-30:
Hi, i'm quite new with octoprint and i'm trying to install octolapse, tried with a fresh reinstall  or my current (in use), same issue.
Isnt octolaps compatible with octo4A?
I'm using a Mi Mix 2, stock pfirmware and no root.

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.7.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Trying to install Octolaps on Octo4a

#### What should have happened?
Worked?

#### What happened instead?
It returns an error

full log:

Installing plugin "Octolapse" from https://github.com/FormerLurker/Octolapse/archive/master.zip...
/usr/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmp7uswdyoz/Octolapse-master.zip --no-cache-dir
Processing /tmp/tmp7uswdyoz/Octolapse-master.zip
Requirement already satisfied: pillow>=6.2.0<7.0.0 in /usr/lib/python3.9/site-packages (from Octolapse==0.4.1) (8.2.0)
Requirement already satisfied: sarge in /usr/lib/python3.9/site-packages (from Octolapse==0.4.1) (0.1.6)
Requirement already satisfied: six in /usr/lib/python3.9/site-packages (from Octolapse==0.4.1) (1.15.0)
Requirement already satisfied: OctoPrint>1.3.8 in /usr/lib/python3.9/site-packages (from Octolapse==0.4.1) (1.7.2)
Requirement already satisfied: psutil in /usr/lib/python3.9/site-packages (from Octolapse==0.4.1) (5.8.0)
Requirement already satisfied: file_read_backwards in /usr/lib/python3.9/site-packages (from Octolapse==0.4.1) (2.0.0)
Requirement already satisfied: setuptools>=6.0 in /usr/lib/python3.9/site-packages (from Octolapse==0.4.1) (52.0.0)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /usr/lib/python3.9/site-packages (from Octolapse==0.4.1) (1.6.5)
Requirement already satisfied: regex in /usr/lib/python3.9/site-packages (from awesome-slugify<1.7,>=1.6.5->Octolapse==0.4.1) (2021.4.4)
Requirement already satisfied: Unidecode<0.05,>=0.04.14 in /usr/lib/python3.9/site-packages (from awesome-slugify<1.7,>=1.6.5->Octolapse==0.4.1) (0.4.21)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2021.2.23)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2021.10.11)
Requirement already satisfied: OctoPrint-PiSupport>=2021.10.28 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2021.10.28)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.1.1)
Requirement already satisfied: markdown<3.2,>=3.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (3.1.1)
Requirement already satisfied: wrapt<1.13,>=1.12.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.12.1)
Requirement already satisfied: flask<2,>=1.1.4 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.1.4)
Requirement already satisfied: Jinja2<3,>=2.11.3 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2.11.3)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.5.0)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.0.0)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2.0)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.0.1)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.1.0)
Requirement already satisfied: cachelib<0.2,>=0.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.1.1)
Requirement already satisfied: PyYAML<6,>=5.4.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (5.4.1)
Requirement already satisfied: pyserial<4,>=3.4 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (3.5)
Requirement already satisfied: netaddr<0.9,>=0.8 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.8.0)
Requirement already satisfied: watchdog==0.10.4 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.10.4)
Requirement already satisfied: netifaces<1,>=0.11 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.11.0)
Requirement already satisfied: pylru<2,>=1.2 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.2.0)
Requirement already satisfied: pkginfo<2,>=1.7.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.8.1)
Requirement already satisfied: requests<3,>=2.26.0 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2.26.0)
Requirement already satisfied: semantic_version<3,>=2.8.5 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (2.8.5)
Requirement already satisfied: Click<8,>=7.1.2 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (7.1.2)
Requirement already satisfied: future<1,>=0.18.2 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.18.2)
Requirement already satisfied: websocket-client<1,>=0.59 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (0.59.0)
Requirement already satisfied: emoji<2,>=1.4.2 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.6.1)
Requirement already satisfied: sentry-sdk<2,>=1.3.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.5.0)
Requirement already satisfied: filetype<2,>=1.0.7 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Octolapse==0.4.1) (1.0.8)
[... 570 lines cut ...]
creating build/temp.linux-aarch64-3.9/octoprint_octolapse
creating build/temp.linux-aarch64-3.9/octoprint_octolapse/data
creating build/temp.linux-aarch64-3.9/octoprint_octolapse/data/lib
creating build/temp.linux-aarch64-3.9/octoprint_octolapse/data/lib/c
gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -g -fno-semantic-interposition -g -fno-semantic-interposition -g -fno-semantic-interposition -DTHREAD_STACK_SIZE=0x100000 -fPIC -I/usr/include/python3.9 -c octoprint_octolapse/data/lib/c/extruder.cpp -o build/temp.linux-aarch64-3.9/octoprint_octolapse/data/lib/c/extruder.o -O3 -std=c++11
error: command 'gcc' failed: No such file or directory
----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-oxycfpro/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-oxycfpro/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-pvt5z1ys/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/Octolapse Check the logs for full command output.
Running setup.py install for Octolapse: finished with status 'error'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

#### Operating System running OctoPrint and Octolapse

OS Name: Android
Os Version: 10

#### Printer model & used firmware incl. version

Printer Model: Ender 5 plus

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: windows

FormerLurker, 2021-11-30:
> error: command 'gcc' failed: No such file or directory

You need to install gcc.  A quick google search will reveal the necessary commands.  You also need the python dev package, if you haven't installed that already.

FYI, you will need gcc to install many things, so it will come in handy.

mawoka-myblock, 2021-11-30:
At first, please wrap the logs into code-blocks and secondly, it doesn't seem to be an issue with OctoLapse, more like an isse with Octo4a since it is complaining about a missing C-compiler which seems to be missing in Octo4a. I don't know if it just doesn't work on Octo4a or if it isn't in your `PATH` or if Octo4a completely misses it.

nandobarreto, 2021-11-30:
> At first, please wrap the logs into code-blocks 

Sorry, i'm new to github as well, i'll try to do this way next time.

> and secondly, it doesn't seem to be an issue with OctoLapse, more like an isse with Octo4a since it is complaining about a missing C-compiler which seems to be missing in Octo4a. I don't know if it just doesn't work on Octo4a or if it isn't in your `PATH` or if Octo4a completely misses it.

Ok, i'll see if it's possible to install GCC on my android. Thank you!

nandobarreto, 2021-11-30:
> > error: command 'gcc' failed: No such file or directory
> 
> You need to install gcc. A quick google search will reveal the necessary commands. You also need the python dev package, if you haven't installed that already.
> 
> FYI, you will need gcc to install many things, so it will come in handy.

Thank you, i'll try to do that.

nandobarreto, 2021-12-05:
> You're getting this still, or something different:
> 
> > error: command 'gcc' failed: No such file or directory
> 
> you might also have to install g++, btw:
> 
> `sudo apt-get install g++`
> 
> If you get other compilation errors (and it's possible, I've never attempted to build for android), let me know and I'll see what I can do. I probably have an android device around here somewhere.
> 
> Let me know.

Well, i tried many different ways to install G++, but doing this in an android without root seems to be impossible.
i'll buy a Pi, will be easier. 

Thanks anyway.

Agwebberley, 2022-02-08:
For all of those in the future that have the same problem, it is possible, all you have to do is tap on the "install plugin extras" button in the settings menu.

nandobarreto, 2022-02-08:
> For all of those in the future that have the same problem, it is possible, all you have to do is tap on the "install plugin extras" button in the settings menu.

what you mean? couldnt find this option

Agwebberley, 2022-02-10:
> 

In the app, in the settings menu, there is an option called "install plugin extras"

nandobarreto, 2022-02-10:
> > 
> 
> In the app, in the settings menu, there is an option called "install plugin extras"

After my question I checked in the app and found it...
and worked like a charm!

for those who want to try, this is your solution.

Dumpster101, 2022-06-11:
I am also having this problem. Not very familiar (yet) with this setup. I can't find the option to "install plugin extras". I'm using Octo A on a fire tablet. Everything works great except for OctoLapse. OctoLapse worked on the first install but I reset the fire tablet to factory default, reloaded everything, and this time OctoLapse errors on install. Can anyone be of help??

MikeOxlong83, 2022-06-22:
I'm also having this problem. OctoPrint intalled on my older Mac, can't install octolapse :(

Chopsticks07, 2023-04-26:
Also struggling to install Octolapse on Pixel4A using Octo4a. Have "Install Plugins" enabled on the app and have tried installing using SSH. 

`Installing plugin "Octolapse" from https://github.com/FormerLurker/Octolapse/archive/master.zip...
/usr/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpfdd19qlo/Octolapse-master.zip --no-cache-dir --use-pep517
Processing /tmp/tmpfdd19qlo/Octolapse-master.zip
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
error: subprocess-exited-with-error
× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> [20 lines of output]
likely tags: v0.4.2
picking v0.4.2
got version from expanded keyword {'version': 'v0.4.2', 'full-revisionid': 'c6c86782cdbfc3590fa4b1bf2e63064eb459b16b', 'dirty': False, 'error': None, 'date': '2023-04-02T10:23:21-0500'}
Traceback (most recent call last):
File "/usr/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
main()
File "/usr/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
json_out['return_val'] = hook(**hook_input['kwargs'])
File "/usr/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
return hook(config_settings)
File "/tmp/pip-build-env-jv2nxdd1/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 341, in get_requires_for_build_wheel
return self._get_build_requires(config_settings, requirements=['wheel'])
File "/tmp/pip-build-env-jv2nxdd1/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 323, in _get_build_requires
self.run_setup()
File "/tmp/pip-build-env-jv2nxdd1/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 487, in run_setup
super(_BuildMetaLegacyBackend,
File "/tmp/pip-build-env-jv2nxdd1/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 338, in run_setup
exec(code, locals())
File "<string>", line 57, in <module>
ModuleNotFoundError: No module named 'octoprint'
[end of output]
note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error
× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.
note: This error originates from a subprocess, and is likely not a problem with pip.
Getting requirements to build wheel: finished with status 'error'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output`