FormerLurker/Octolapse #798: Unable to rende

pinoystyle, 2022-01-04:
Since today my videos wont render anymore.

I get the error message Octolapse Rendering Failed
Rendering failed for camera 'Webcam - Default OctoPi 0.16.0'. render-error: Unknown render error. Please check plugin_octolapse.log for more details.. Inner Exception: not enough arguments for format string

Log is attached

#### Version of Octolapse
Octolapse Version: 0.4.1

#### Version of OctoPrint
OctoPrint Version: 1.7.2

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred
I found that Octolapse told me that there are unrendered videos. As soon as i hit render video i got the error

#### What should have happened?
Video should be rendered

#### What happened instead?
I got an error message

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Anycubic i3 Mega S
Printer Firmware Version: Knutwurst Marlin 1.3.1

FormerLurker, 2022-01-04:
Try installing the devel version.  There are issues with some special characters (percent signs) in the file names due to a recent change in Octoprint.  Let me know if you need help trying it out.

bpengu1n, 2022-02-12:
> Try installing the devel version.  There are issues with some special characters (percent signs) in the file names due to a recent change in Octoprint.  Let me know if you need help trying it out.

I'm having the same issue, and still get it with the devel version :(

Priscil87, 2022-03-30:
I have the same issues, devel version didn't help and still the same error message

pinoystyle, 2022-03-30:
i resolved it for me by having no special characters in my g-code filename

Priscil87, 2022-03-30:
> i resolved it for me by having no special characters in my g-code filename

I've been looking for a solution for weeks and it can be that simple. Thanks!!!!!!

msoffen, 2022-06-13:
Is there anyway to process the downloaded JPG images on a Linux system?