FormerLurker/Octolapse #799: Will not Print ...

jseaz73, 2022-01-05:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 1.7.2

#### Version of OctoPrint

OctoPrint Version: ___REPLACE_THIS__OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Load Gcode ...get error about missing Printer config
2. CAN not print because of error

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: _Windows
Os Version:  10

#### Printer model & used firmware incl. version

Printer Model: Ender 3 Pro
Printer Firmware Version:  Marlin (build date 20211215)

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.yout
[CE3E3SE3_Light_holder_-_Left.gcode2.zip](https://github.com/FormerLurker/Octolapse/files/7818119/CE3E3SE3_Light_holder_-_Left.gcode2.zip)
ube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).
[CE3E3SE3_Light_holder_-_Left.gcode2.zip](https://github.com/FormerLurker/Octolapse/files/7818122/CE3E3SE3_Light_holder_-_Left.gcode2.zip)

jseaz73, 2022-01-05:
Octolapse Startup Failed
Slicer Settings Not Found
No slicer settings were not found in your gcode file.

FormerLurker, 2022-01-07:
You have automatic slicer settings detection enabled, which is the recommended setting, but I'm guessing you are using Cura and haven't added the necessary script to your start gcode.  Please [take a look at this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Automatic-Slicer-Configuration), and follow the appropriate steps for your slicer.  Let me know if that helps.

jseaz73, 2022-01-08:
I followed those steps and have them in the before G code per the
instructions.

jseaz73, 2022-01-08:
I attached my GCode ...

FormerLurker, 2022-01-08:
Here is the filled script I found in your gcode:

[image]

However, the comment markers ';' were stripped.  Here's what it should look like in your endgcode (though it would be best to put it in your start gcode, as it will be faster for Octolapse to find):

```
; Script based on an original created by tjjfvi (https://github.com/tjjfvi)
; An up-to-date version of the tjjfvi's original script can be found
; here:  https://csi.t6.fyi/
; Note - This script will only work in Cura V4.2 and above!
; --- Global Settings
; layer_height = {layer_height}
; smooth_spiralized_contours = {smooth_spiralized_contours}
; magic_mesh_surface_mode = {magic_mesh_surface_mode}
; machine_extruder_count = {machine_extruder_count}
; --- Single Extruder Settings
; speed_z_hop = {speed_z_hop}
; retraction_amount = {retraction_amount}
; retraction_hop = {retraction_hop}
; retraction_hop_enabled = {retraction_hop_enabled}
; retraction_enable = {retraction_enable}
; retraction_speed = {retraction_speed}
; retraction_retract_speed = {retraction_retract_speed}
; retraction_prime_speed = {retraction_prime_speed}
; speed_travel = {speed_travel}
```

Notice every line starts with ```; ```.  That is important.

jseaz73, 2022-01-08:
AHHHH I thought that the Comments markers ';' needed to be removed to be
processed :-(

Thank you

If that was in the instructions, it apologize for missing them ..

I am well versed in JavaScript and a couple of other languages but VERY new
to gcode :-)

thank you,

Jason

FormerLurker, 2022-01-08:
> If that was in the instructions, it apologize for missing them ..

Not a problem.  Us devs have a tendency to read our own experience into programming related topics, so I totally understand.

> I am well versed in JavaScript and a couple of other languages but VERY new to gcode :-)

Honestly, Octolapse is a good place to start.  Unfortunately, as with lots of programming related topics, there is no real standard to follow, so Octolapse needs a lot of info to do things correctly.  Learning gcode, and firmware specific implementations (like the dreaded G90/G91 influences extruder setting) will go a long way towards ensuring you get the prints you want, and understanding why things go wrong.  In this case, it's just a design decision on my part.  The slicer settings extractor first tries to figure out what slicer you are using, then based on that it looks for comments that include the required settings.  This varies from slicer to slicer, but I based this decision on the slicers that DO provide their settings in the gcode by default.  Unfortunately cura is not one of those slicers.  I actually added a feature request for this, but Ultimaker thinks it just clutters the gcode.  Don't even get me started on multi-extruder setups, lol!

Valencanoextinto, 2022-01-20:
Hi i have got de similar problem 
i'm use cura 4.13 and when i send to print my stl i have got this problem 
[image]

thanks

jseaz73, 2022-01-20:
Did you make sure to add the code shown above (the second code window) to your printer START GCODE block ?

and KEEP the ';' in front of the code

Valencanoextinto, 2022-01-21:
thanks for the answer 
yeah i am sure this is my gcode of my printer
[image]

the problem is that i have been using octolapse since last year whitout problems.
this error has been appear 2 days ago 

thank for helping me and have a nice weekend