FormerLurker/Octolapse #800: Sony Alpha series, long pause after snapshot

jeffro1265, 2022-01-08:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.7.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

Attempting to take a time-lapse with Sony Alpha A5000 and A6400. 

#### What should have happened?
Print head moves to stabilization location, photo is taken immediately, print head immediately moves back to work piece. 

#### What happened instead?
Print head moves to stabilization location, immediately takes photo, pauses for 5-6 seconds, then resumes print. 

#### Operating System running OctoPrint and Octolapse

OS Name: ___Octopi 0.18.0 - Raspberry Pi 3B
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___Creality CR-6 SE
Printer Firmware Version: CFW 6.0

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://we.tl/t-iQKhkLzQpN sliced with cura 4.12.1

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  (https://we.tl/t-Mc7YnggSiD)

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://we.tl/t-EDjJ1zKTeR

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2022-01-08:
Yeah, sorry, the Alpha is a huge pain in the neck regarding remote control.  Your best bet is to reduce your image size as much as possible (turn down resolution as much as you can, increase jpg compression as much as you can) to reduce the image acquisition time (time spent downloading the image across the usb cable).

As far as I know, the alpha does not support the two more advanced forms of image acquisition that I have guides for (see the wiki): No download (images are stored only on the camera's SD card), and deferred download (images are all downloaded to the PI AFTER the print is completed), so there isn't much alternative.

Maybe you could post this issue to the GPhoto2 repository?  I'm not sure if there have been any recent developments related to this camera.  It is quite possible that they have done everything possible already, but maybe newer firmware offers more capabilities?  

Just FYI, I do NOT maintain, and do not really have anything to do with the most excellent gphoto2 library, but I know the author is quite responsive.  If you do post on their git issues, please be extra kind to this open source dev for creating such a wonderful tool!

Thanks

jeffro1265, 2022-01-08:
Wow  man thanks for the reply. I see your work all over and its incredible. The dedication you have to keeping this function alive is admirable. 

I have spent the last 2 weeks scouring countless forums and subreddits looking for answers any nothing seems to work 'Correctly'. I've tried the save to SD only method but you are correct in that it does not detect the storage mediums and therefor does not work. Ive tried adjusting camera settings endlessly with little to no success. Ive tried faster SD cards and even the latest and greatest USB boot method. 

In looking at the log, it appears that the main cause for delay (5.8 seconds) is due to post processing, presumably on the Octoprint side. Ive tried and tried to get it to not render the images on the Pi with no success. 

>TypeError: expected str, bytes or os.PathLike object, not NoneType
>2022-01-08 14:27:53,120 - octolapse.settings - ERROR - The rendering settings file does not exist for the given snapshot job at /home/pi/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp/0edfbe31-de3b-48f1-b726-7344dd8db429/39590c20-ddd1-4af3-bf2a-fea41515296f/rendering_settings.json.
>2022-01-08 14:27:53,123 - octolapse.settings - ERROR - The camera settings file does not exist for the given snapshot job at /home/pi/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp/0edfbe31-de3b-48f1-b726-7344dd8db429/39590c20-ddd1-4af3-bf2a-fea41515296f/camera_settings.json.
>2022-01-08 14:31:02,758 - octolapse.snapshot - INFO - Starting snapshot acquisition
>2022-01-08 14:31:02,762 - octolapse.snapshot - INFO - Starting 1 snapshot threads.
>2022-01-08 14:31:02,765 - octolapse.snapshot - INFO - Snapshot - running snapshot script for the A6400  camera.
>2022-01-08 14:31:02,767 - octolapse.snapshot - DEBUG - Snapshot Delay - No pre snapshot delay configured for the A6400  camera.
>2022-01-08 14:31:02,768 - octolapse.script - DEBUG - Executing A6400  - Snapshot Camera Script with no timeout: "/home/pi/scripts/take-snapshot.sh" "0" "0.0" "/home/pi/.octoprint/data/octolapse/tmp" "/home/pi/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp/99188481-ca07-49f4-81fe-69834df2589a/39590c20-ddd1-4af3-bf2a-fea41515296f" "squirrel tiny000000.jpg" "/home/pi/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp/99188481-ca07-49f4-81fe-69834df2589a/39590c20-ddd1-4af3-bf2a-fea41515296f/squirrel tiny000000.jpg" "A6400 "
>2022-01-08 14:31:08,618 - octolapse.script - DEBUG - Console output (stdout) for 'A6400  - Snapshot Camera Script':Creating directory: /home/pi/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp/99188481-ca07-49f4-81fe-69834df2589a/39590c20-ddd1-4af3-bf2a-fea41515296f
>Model                          Port            
>----------------------------------------------------------
>USB PTP Class Camera           usb:001,004     
>New file is in location /capt0000.jpg on the camera
>Saving file as /home/pi/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp/99188481-ca07-49f4-81fe-69834df2589a/39590c20-ddd1-4af3-bf2a-fea41515296f/squirrel tiny000000.jpg
>Deleting file /capt0000.jpg on the camera
>2022-01-08 14:31:08,621 - octolapse.snapshot - DEBUG - Post-processing snapshot for the A6400  camera.
>2022-01-08 14:31:08,622 - octolapse.snapshot - INFO - The snapshot script job completed, signaling task queue.
>2022-01-08 14:31:08,625 - octolapse.snapshot - INFO - Snapshot threads complete, but may be post-processing.
>2022-01-08 14:31:08,626 - octolapse.snapshot - INFO - Snapshot acquisition completed in 5.865 seconds.
>2022-01-08 14:31:09,129 - octolapse.snapshot - DEBUG - Post-processing snapshot for the A6400  camera complete.

As you mentioned about gphoto, for some reason i cannot get the dev version to install correctly. When i select option 1 to install the development version, then run the whereis gphoto2, it he query is blank. I assume that means it did not install. Same for libgphoto2. So im stuck with: 

>This version of gphoto2 is using the following software versions and options:
>gphoto2         2.5.20         gcc, popt(m), exif, cdk, aa, jpeg, readline
>libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
>libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

Ill keep trying to get gphoto to update as their latest changelog indicates speed ups. but im not the greatest with linux so its gonna take some time. 

Appreciate what you do and ill be buying you a "coffee" for your efforts.

FormerLurker, 2022-01-08:
Are you positive that you are saving only a jpg, and not a .raw file?  Octolapse will attempt to convert .raw to . jpg, if possible, but these files are huge, and time consuming to process.

In any case, please keep me posted about your efforts.  The alpha series has been a thorn in my side for a long while, which is a shame, because it is an otherwise great camera!

About the install script, that is a port from another GitHub user (see the citation in the script).  Maybe there is a newer and better version available?  All I did to the original script was to modify the default git location, which was changed in octopi for various reasons.

jeffro1265, 2022-01-08:
Oh my lord, it’s working properly and without delay on the Sony alpha A5000. On the go currently but will update this when I return on what fixed it. There is going to be a lot of happy people out there.

FormerLurker, 2022-01-08:
Fan friggin tastic!  Consider writing your follow up as a guide, and I will add it to the dslr guide notes.  Thanks!

jeffro1265, 2022-01-09:
The issue was with the version of gphoto2 and libgphoto2 that was being installed from the script in the "configuring a DSLR" guide. Option 1 of that script calls to install the latest dev version, but when executed, it doesn't install it correctly resulting in an empty whereis query for gphoto2. Option 2 of the script in the guide installs correctly, but only pulls down gphoto2 version 2.5.20 as indicated above. No matter what i tried, i couldn't force an update for some reason. 

As you mentioned, you pulled the script for somewhere else and advise that i go directly to the source for a more up-to-date gphoto2 install. After going to this link ( https://github.com/gonzalo/gphoto2-updater ) i used the following command to uninstall the old versions of gphoto2 and libgphoto2:

>sudo apt-get remove gphoto2 libgphoto2-6 libgphoto2-dev libgphoto2-l10n libgphoto2-port12

then reinstalled using the updater script at the link above:

>wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh && chmod +x gphoto2-updater.sh && sudo ./gphoto2-updater.sh

This new install command installed both gphoto2 and libgphoto2 to the latest development versions. 2.5.28.1. I did have to update the visudo file as the install directory was different than what the script in your guide installs to. I used whereis to determine the new install location.

Looking through the changelogs for gphoto2, they include some comments about speeding up the processes required to trigger and capture photos and that's exactly what it did for me. 

I have tested this on both the Sony A5000 and Sony A6400  now. In conjunction with Octolapse, my A5000 and A6400 both move to the stabilization point, snaps a photo to the SD and OctoPi, and quickly resumes printing. See video below of it functioning properly. (print is small for testing purposes)  https://youtu.be/4rZNu8U8aYU. For reference, the older version of gphoto2 i had installed would snap the photo then pause for a solid 6 seconds before resuming the print. This resulted in many hours of additional print time on larger prints. Video of how it used to be, https://youtu.be/GxazDJvcq2g , FRUSTRATING!

Again, i thank you for your work on this. One simple step was causing the headache. I am not a linux person so this could have been a walk in the park for some.

My suggestion regarding your guide would be to have users use the command above to install gphoto. Given the volume of people looking for fixes for the alpha series, i think it would be beneficial. 

Regards,

donnysnarko, 2022-01-28:
Hello all (@FormerLurker & @jeffro1265 ), ive been trying to troubleshoot this again folowing the success here (i.e getting Octolapse to work on my Sony A6500), but even after following the suggestion to use the dev. version of gphoto2 I still havent had success.  
 
1. The issue i'm havingis perhaps unrelated to the version of gphoto2 i was using since im getting the same issue as all  my previous attempts, which is that the camera is able to take a picture, but then will only save it in a location called **/capt0000.arw** which is a non-location on the camera (i.e. doesnt appear to exist anywhere on the camera - i.e. it presumably never saves). - [Here is what I mean](https://imgur.com/a/bSJpi46)

2. Its the same issue that I posted about [here](https://imgur.com/a/bSJpi46) last year, to which it was suggested that i try **gphoto2 --set-config d222=17 --capture-image**, which i did on the recently updated dev version and got[ this response](https://imgur.com/a/jQ9U0et) - again not working, but a lisghtly different system response than before.

3. The SD is formatted to EXT4 and then reformated in-camera to ExFat (not sure if this is relevant). 
4. I have the USB connection set to PC remote, which im guessing is correct since the gphoto2 --auto-detect identified the camera and model that im using.  

I havent even got as far as trying to use this with the Octolapse software, im still struggling with getting gphoto2 to take a picture and save it SD card (in-camera)..any help on what i may be doing wrong would be awesome, since it feels like it;s a rookie error im making.

*Working with the 2.5.28.1 version of Gphoto2.*