FormerLurker/Octolapse #801: OctoEverywhere Bug?

QuinnDamerell, 2022-01-11:
Hello!

I'm the developer behind OctoEverywhere! I had a member of my common reach out to me saying they were having an issue and they thought it might be related to using Octolapse + OctoEverywhere. If there are any issues please let me know and I would love to fix them! OctoEverywhere should be 100% transparent to any plug-in, if it's not, I want to fix it!

Thanks and thank you for your amazing plug-in!

FormerLurker, 2022-01-12:
I haven't seen this reported issue with my own eyes.  I will have to give your plugin a go and see for myself, especially since you took the time to send me a message!  Will let you know what I find, but it will be a couple weeks.  I am finishing a huge rush job for a client.

Thanks for your kind words!  Am looking forward to seeing what your plugin is all about!

QuinnDamerell, 2022-01-12:
@FormerLurker sounds great! I'm also happy to put some time into debugging it if you would like, just send me the details.

Going forward as well, if you ever get a bug or issue related to OctoEverywhere please feel free to shoot it over to me. You can open a bug in my [plugin repo](https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere), tag me in your repo, or open a [support ticket](https://octoeverywhere.com/support) (which will go directly to me.)

FormerLurker, 2022-01-12:
I appreciate the offer!  I will definitely take you up on it, though I suspect this issue might be setup related.  We will see :)