FormerLurker/Octolapse #810: Printing is not starting

RustyWraith, 2022-01-19:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: v0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.7.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Start print

#### What should have happened?
Printer should home.

#### What happened instead?
Nothing.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Custom build delta
Printer Firmware Version: klipper v0.10.0-221-g8a3727ef

#### Browser and version of browser, operating system running browser

Browser: Chrome Version 97.0.4692.71 (Official Build) (64-bit)
Browser OS: win 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: [Bolt.gcode.zip](https://github.com/FormerLurker/Octolapse/files/7899532/Bolt.gcode.zip)

#### Link to settings.json

Link to settings.json with all passwords removed: 
[config.zip](https://github.com/FormerLurker/Octolapse/files/7899633/config.zip)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  [plugin_octolapse (18).log](https://github.com/FormerLurker/Octolapse/files/7899602/plugin_octolapse.18.log)

#### Link to octoprint.log

Link to octoprint.log:  [octoprint (1).log](https://github.com/FormerLurker/Octolapse/files/7899621/octoprint.1.log)

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

RustyWraith, 2022-01-22:
RPi4.
Connected to https://community.octoprint.org/t/octoprint-hanging-on-print-starting/23841/26

FormerLurker, 2022-04-26:
So sorry it's taken so long to get back with you, but I think this is related to the python version.  If you are willing to try out the devel branch, this may be solved.  Please let me know, thanks!

RustyWraith, 2022-04-26:
Hello!
No luck with devel branch.
[image]
[octoprint (2).log](https://github.com/FormerLurker/Octolapse/files/8564638/octoprint.2.log)
[plugin_klipper_debug.log](https://github.com/FormerLurker/Octolapse/files/8564639/plugin_klipper_debug.log)
[plugin_octolapse (19).log](https://github.com/FormerLurker/Octolapse/files/8564640/plugin_octolapse.19.log)

FormerLurker, 2022-04-26:
Thanks!  The default 10 second startup timeout is definitely getting hit, which I've confirmed from your serial log.  I will need to add some additional logging to figure out what's happening.  I don't have enough logging apparently to figure out where things are going wrong.  The last useful message I see is that the __init__.on_print_start function.  I'll add some logging here and will push to devel ASAP.

RustyWraith, 2022-04-26:
Thank you! I'll be watching for commits.

FormerLurker, 2022-04-26:
Hey, as a quick test, I assume you are using the smart triggers?  you could try one  of the classic layer change triggers and see if things start working.  That will eliminate a whole section of the program (as well as some quite complicated extensions).  In the meanwhile, I'll be adding logging as time permits.

FormerLurker, 2022-04-26:
I have another idea I ran into while adding some logging:  Maybe try removing the klipper macros that are in your start gcode for now?  Not sure how that will affect your probe, but these specifically:

SET_PRESSURE_ADVANCE ADVANCE=0.25
SET_PIN PIN=z_prb VALUE=1
DRY_PROBE_BED
PROBE_BED

At a minimum, move 'SET_PRESSURE_ADVANCE' below the warmup commands like so:

G28
M190 S75
M109 S235
SET_PRESSURE_ADVANCE ADVANCE=0.25
SET_PIN PIN=z_prb VALUE=1
DRY_PROBE_BED
PROBE_BED

These could confuse Octolapse, but I need to run some tests.

RustyWraith, 2022-04-26:
I think, I've done this some time ago. Will try again.

RustyWraith, 2022-04-26:
No luck there.

FormerLurker, 2022-04-26:
K, just pushed some preliminary logging commands to the 'devel' branch.  Please make sure you use the 'log everything' option to capture all of the 'verbose' log entries.  If we need additional logging, I'll add more until we figure out where it is stalling and why.

RustyWraith, 2022-04-26:
Something like that.
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/8565939/octoprint.log)
[plugin_klipper_debug.log](https://github.com/FormerLurker/Octolapse/files/8565940/plugin_klipper_debug.log)
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/8565941/plugin_octolapse.log)
[serial.log](https://github.com/FormerLurker/Octolapse/files/8565942/serial.log)

FormerLurker, 2022-04-26:
Interesting, it's hanging testing out the webcam.  I did NOT expect that!

First, can you export and attach all of your webcam (dslr won't be a problem here) profiles:

[image]

Then, try the 'Test Webcam' button and just see what happens:

[image]

Might need to add more logging to the camera test script, but I'll wait for your response.

RustyWraith, 2022-04-26:
"Test Webcam" leads to octoprint to hang until power reset.
I use an old phone with "IP Cam" app over wifi.
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/8566020/octoprint.log)
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/8566022/plugin_octolapse.log)
[Camera_Profile.json.txt](https://github.com/FormerLurker/Octolapse/files/8566025/Camera_Profile.json.txt)

FormerLurker, 2022-04-26:
Ok, I think I've figured it out.  The camera address in your profile, unless I am mistaken, is for a video stream, and not a snapshot:

http://192.168.1.188:8080/video

If you can provide information about the app you are using, or if you can search yourself to see if there is an address that returns a jpg image instead of a stream, that should solve the problem.

It also appears as though the app is not providing an appropriate mime type (or there is a bug in Octolapse detecting this, which is entirely possible), and Octolapse is basically downloading forever, since there is no 'end' to the stream.  This would result in a lockup, and finally a memory overflow error, possibly followed by a crash.

Even if we get this fixed, please hold onto that profile you saved, and we can retest if/when I can figure out how to detect and prevent this issue.  The standard timout won't work, because the http connection is working and actively downloading.  I might have to add some second timeout to catch and report the error.

One thing you could do to help me would be to record the header information in the browser's debugger when opening a stream to that camera.  Something in there could, in theory, be used to detect this issue without hanging forever.  If not, I can probably replicate it myself by installing the app you are using.

RustyWraith, 2022-04-26:
Now it's working!
I use this app.
ip:8080 is a web interface
ip:8080/video/ is a stream
ip:8080/photo.jpg/ is a snapshot
https://play.google.com/store/apps/details?id=com.pas.webcam&hl=en&gl=US
Headers are looking like that?
[image]
[image]

FormerLurker, 2022-04-26:
Awesome!  

Based on your headers, I've set this test to use streaming, and added a timeout to the loop.  Ive pushed to the devel branch, so maybe you can test with the old url and see if it correctly shows this message:  A read timeout occurred while connecting to {0} for the '{1}' camera profile.  Are you using the video stream URL by mistake?

If it works, I will consider updating all of the requests, so that I control connection, read, and total session duration timeouts.  Currently only 'Connection' and 'Read' are supported.  Strangely 'Read' only timeouts if there are delays between requests, but I suppose that would be useful in most cases.

Note that {0} and {1} are tokens, and will be replaced with the URL and profile name respectively.

Post your first octolapse, I'd love to see it, and looking forward to see if you get a sensible error message when using a streaming url!

RustyWraith, 2022-04-26:
Unfortunately, I was using video stream by avoiding RTFM rule.
The error raises as expected.
[image]

FormerLurker, 2022-04-26:
> Unfortunately, I was using video stream by avoiding RTFM rule.

Lol!  How else would we learn :)

> The error raises as expected.

Excellent!  I realize now that I need to apply the retry rule still, but this should be adequate for all but the most obscure case scenarios.

Going to mark this as closed, but feel free to drop a link to your first timelapse, and enjoy!

closed 2022-04-26 by FormerLurker