FormerLurker/Octolapse #815: Could not render movie, got return code -11:

rpoland007, 2022-02-04:
Hello, my octolapse just stopped rendering movies, and now I get this error code (Log below)

Is there a location where the error codes and their explanations is listed? anyone see this error before?

Thanks, Robert

2022-02-03 13:12:29,524 - octolapse.render - ERROR - Rendering Error
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/render.py", line 1591, in _render
    return_code, stderr_text))
octoprint_octolapse.render.RenderError: return-code: Could not render movie, got return code -11: 
2022-02-03 14:07:08,841 - octolapse.__init__ - INFO - Started logging to file.
2022-02-03 14:07:08,870 - octolapse.settings - INFO - Loading existing settings file from: /home/pi/.octoprint/data/octolapse/settings.json.
2022-02-03 14:07:08,883 - octolapse.settings - INFO - Creating settings from iterable.
2022-02-03 14:07:08,947 - octolapse.settings - INFO - Settings created from iterable.
2022-02-03 14:07:08,954 - octolapse.settings - INFO - Settings file loaded.
2022-02-03 14:14:45,174 - octolapse.__init__ - INFO - Started logging to file.
2022-02-03 14:14:45,179 - octolapse.settings - INFO - Loading existing settings file from: /home/pi/.octoprint/data/octolapse/settings.json.
2022-02-03 14:14:45,188 - octolapse.settings - INFO - Creating settings from iterable.
2022-02-03 14:14:45,217 - octolapse.settings - INFO - Settings created from iterable.
2022-02-03 14:14:45,222 - octolapse.settings - INFO - Settings file loaded.
2022-02-03 14:18:18,680 - octolapse.render - ERROR - Rendering Error
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/render.py", line 1591, in _render
    return_code, stderr_text))

FormerLurker, 2022-04-26:
Did you recently upgrade python?  If so, I think I know what's going on.  Let me know please, and sorry for the unacceptably long delay getting back to you.  Been out of action for reasons beyond my control.

rpoland007, 2022-04-26:
Hello,

I did get it resolved, but it was on my end, my SD card failed shortly after this and everything has been working normally and well since then.

Thanks for getting back to me too, much appreciated.

Have a good day,

-Robert

FormerLurker, 2022-04-26:
An SD failure makes sense, THANK YOU for getting back to me much faster than I got back to you!  Please don't hesitate to post again if you have other issues/suggestions.  Marking this closed!

closed 2022-04-26 by FormerLurker