FormerLurker/Octolapse #816: [Request] Add setting to take multiple snapshots per layer

ansonl, 2022-02-06:
#### Add setting to take multiple snapshots per layer

Some of my prints are topo maps which are very wide but only 30-40 layers tall. Currently 1 snapshot per layer produces a very short timelapse with the printhead jumping everywhere in "printing" positioning mode due the way some topo maps have individual peaks. If there was an option to increase the number of snapshots taken per layer, users with wide, low layer prints would be able to make longer timelapses that appear smoother and the "printing" positioning might be able to select better locations for snapshots with more chances per layer. 

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.7.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2022-04-26:
You can use the 'Timer' trigger.  This is what I use when I have a very short print.  It has an interesting effect.

FYI, I'm planning to add a 'length' trigger that would take a snapshot after so many mm of filament is extruded.  This has the benefit of working with the new 'smart' trigger types, and would also solve your issue.  Look for this in the next version of Octolapse.

ansonl, 2022-04-28:
Thanks, I'm in the middle of a printer rebuild and will try out the Timer trigger once everything is back together!

ansonl, 2023-03-01:
@FormerLurker I tried your suggestion to try the Timer trigger and realized that the constant timer between snapshots favors screen time for layers that take longer to print (naturally). I was actually wondering if Octolapse is able to take a set number of snapshots evenly spread out at each layer so every layer gets the same number of snaps (ex: 5) and the final render will appear as if the print "progressing" up through the layers at constant speed. 

For example if a model has a wide bottom, narrower middle, and wider top looking like below from the side:

```
    |     _________
    |     \           /
    |       |         |
Z  |     /           \
    |   /__________\
    | ----------------
              X
```

With a timer, the bottom and top layers will appear slower in the timelapse render because more snapshots are being taken of them (because longer time to print bigger layer). And the timelapse will appear to "speed up" as the middle is printed and slow down back towards the top. 

The Layer trigger favors each layer evenly so the model appears to print at a constant speed vs height as each layer gets one snap. However, I'm printing lower, flatter models which don't have many layers (about 20-50) and 1 snapshot per layer makes for too short of a timelapse. Since Octolapse already preprocesses the file, it would be great if a constant number of snapshots spaced out within each layer was available and I think it would make for a cool new effect with minimal processing.