FormerLurker/Octolapse #820: Not showing timelapse preview

Lucasroemerisaksen, 2022-02-08:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: ___0.4.1

#### Version of OctoPrint

OctoPrint Version: ___1.7.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___NO

#### What were you doing when the problem occurred

1. I clicked on the timelapse button on octoprint
2. Scrolled down to the bottom of the page to the already rendered timelapses
3. Clicked on the show / view / play button shaped as a camera, it did not load, just gray image not timelapse.

#### What should have happened?
it should have shown the timelapse that was rendered and clicked on

#### What happened instead?
it does not play the timelapse, just shows a short loading circle, and then grey screen where the timelapse should be.

#### Operating System running OctoPrint and Octolapse

OS Name: ___windows
Os Version: ___windows 10 and 11

#### Printer model & used firmware incl. version

Printer Model: Ender 3 pro
Printer Firmware Version: Marlin, version unknown

#### Browser and version of browser, operating system running browser

Browser: ___chrome
Browser OS: ___windows

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2022-04-26:
Sorry for the delay in responding.  I've been out of action for a while, and am finally able to get back to my various open source projects.  

Can to take a few screenshots showing what you are seeing?  Also, it might be good to open the developer tools in your browser (typically F12), and search the console for errors.

lin-ycv, 2022-11-04:
I'm also experiencing this issue where the video won't render in the timelapse tab, but will play if I download it
[image]

Haagimus, 2023-02-11:
@FormerLurker I also am having the same issue as these folks. I am getting an HTTP error 304 when trying to play the video in the player. Downloading it works fine.

FormerLurker, 2023-04-02:
So, I will see what I can do about this, but I am not the code owner for the built-in timelapse preview.  My guess is that the format or options for the rendering profile are not able to be previewed using the settings Octolapse uses when rendering.  I am 100% certain that some of the rendering options will NOT ever work with the preview button, as they are not supported by any browsers.