FormerLurker/Octolapse #826: GPhoto2 with DSLR: "ERROR: Could not get image." - Download from Camera failed

JRundfeldt, 2022-02-26:
#### Version of Octolapse

Octolapse Version: v0.4.1 stable

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.7.3 

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. I used the "Test Script" Button, intending to set up a DSLR (EOS 500D)

#### What should have happened?
I expected the Camera to take a picture and the .CR2 File to be downloaded to the Raspberry Pi

#### What happened instead?
The Camera shot a picture, an error message was shown. The Error message on the UI was: 

Camera Script Failed
Errors were detected - The 'EOS-500D - Snapshot Camera Script' process returned errors. See plugin_octolapse.log for details

No picture was downloaded. 

#### Operating System running OctoPrint and Octolapse

OS Version: 
OctoPi 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Creality CR-10S (Printer was switched off and no serial connection was established during the test - yesterday evening i shot successfully a timelapse, but no pictures were downloaded from the camera)

#### Browser and version of browser, operating system running browser
Browser: Chrome latest
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred
No GCode was used. 

#### plugin_octolapse.log

2022-02-26 12:35:14,564 - octolapse.script - DEBUG - Executing EOS-500D - Snapshot Camera Script: "/home/pi/scripts/take-snapshot.sh" "0" "10" "/tmp/tmpl48wm2be" "/tmp/tmpl48wm2be/dcad89bc-6424-4623-b785-8d22622c2798/0e448b3b-1089-4888-9ff5-747f04e41340" "test_snapshot000000.jpg" "/tmp/tmpl48wm2be/dcad89bc-6424-4623-b785-8d22622c2798/0e448b3b-1089-4888-9ff5-747f04e41340/test_snapshot000000.jpg" "EOS-500D"
2022-02-26 12:35:18,357 - octolapse.script - DEBUG - Console output (stdout) for 'EOS-500D - Snapshot Camera Script':snapshotnumber: 0
	Creating directory: /tmp/tmpl48wm2be/dcad89bc-6424-4623-b785-8d22622c2798/0e448b3b-1089-4888-9ff5-747f04e41340
	Model                          Port            
	----------------------------------------------------------
	Canon EOS 500D                 usb:001,003     
	New file is in location /capt0000.jpg on the camera
	Saving file as /tmp/tmpl48wm2be/dcad89bc-6424-4623-b785-8d22622c2798/0e448b3b-1089-4888-9ff5-747f04e41340/test_snapshot000000.jpg
	Deleting file /capt0000.jpg on the camera
	New file is in location /capt0000.cr2 on the camera
	File /tmp/tmpl48wm2be/dcad89bc-6424-4623-b785-8d22622c2798/0e448b3b-1089-4888-9ff5-747f04e41340/test_snapshot000000.jpg exists. Overwrite? [y|n] 
2022-02-26 12:35:18,358 - octolapse.script - ERROR - Error output (stderr) for 'EOS-500D - Snapshot Camera Script':
	ERROR: Could not get image.

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

No Loglines were logged in this Logfile during reproduction of the issue

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 
Error message that was shown in the WebUI
[image]

JRundfeldt, 2022-04-27:
I am so so sorry, i am super busy at work and havent yet gotten to it to try it. Thanks a ton for your help - i have this issue bookmarked and the moment i have a bit of time, i will try your suggestions

FormerLurker, 2022-04-27:
@JRundfeldt, take your time!  I certainly did getting around to your issue, so fair play :)

JRundfeldt, 2022-04-30:
You were absolutly right - my camera was set to JPEG+RAW. I now tested with JPEG-Setting and with RAW-Setting alone, and it works nicely!

I will try to make a print now and see if that works out too. 

I believe this little check "Make sure to set your DSLR to either RAW or JPEG, but not both at the same time" should be part of the wiki - is there a way I can edit that?

closed 2022-04-30 by JRundfeldt

JRundfeldt, 2022-05-01:
@FormerLurker it works! Thx! https://twitter.com/ijonberlin/status/1520776446544912384