FormerLurker/Octolapse #827: [Request]

mbell90, 2022-02-27:
#### If this is a feature request describe it here

can a profile for the Vyper be added under anycubic.

#### Version of Octolapse
 
Octolapse Version: ___REPLACE_THIS__OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: ___REPLACE_THIS__OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2022-04-26:
I recently received a Vyper, and will be adding a profile.  When I get a test profile created, I'll post it here for you to try.

Sorry for the delay, been gone for a while due to some personal issues.

mbell90, 2022-04-26:
No problem. I honestly forgot I made this request. I just new that loads of
people were buying this and setting up the octopis

FormerLurker, 2022-04-26:
You didn't happen to get a profile working in the meantime, did you?

mbell90, 2022-04-26:
I did. Modified the one for the mega i3 I think.

FormerLurker, 2022-04-26:
Would you be willing to export the profile and attach it?  You can do that
within the octolapse settings (download icon next to the profile).

mbell90, 2022-04-26:
Sure. Not perfect tho. Has a bit of wabble I. The video output. But works.
I'll try and send that tonight or tomorrow

tomlilly, 2023-01-11:
Did a Vyper profile ever emerge from this?

tomlilly, 2023-01-16:
Touching on this again, I modified the I3 Mega profile a bit and have a "working" Vyper profile. I couldn't get it to start correctly at the first layer but other than that it works well enough.  
  
Unable to attach a .json file, so here are the contents for anyone who wants it.

<details>
<summary>AnycubicVyper.json</summary>

`{"version": "0.4.1", "settings_version": "0.4.0", "type": "Printer", "profile": {"name": "Anycubic - I3 Mega", "description": "Anycubic I3 Mega - Version 1.0", "guid": "5b67fbd3-3753-45dd-92db-5278df18e75a", "automatic_configuration": {"key_values": [{"name": "Anycubic", "value": "anycubic"}, {"name": "I3 Mega", "value": "i3_mega"}], "version": "1.2", "suppress_update_notification_version": false, "is_custom": true}, "has_been_saved_by_user": true, "slicer_type": "automatic", "gcode_generation_settings": {"vase_mode": null, "layer_height": null, "extruders": []}, "slicers": {"cura": {"slicer_type": "cura", "version": "unknown", "extruders": [{"version": null, "speed_z_hop": null, "max_feedrate_z_override": null, "retraction_amount": null, "retraction_hop": null, "retraction_hop_enabled": false, "retraction_enable": false, "retraction_speed": null, "retraction_retract_speed": null, "retraction_prime_speed": null, "speed_travel": null}], "axis_speed_display_settings": "mm-sec", "layer_height": null, "smooth_spiralized_contours": false, "machine_extruder_count": 1}, "simplify_3d": {"slicer_type": "simplify-3d", "version": "unknown", "extruders": [{"retraction_distance": null, "retraction_vertical_lift": null, "retraction_speed": null, "extruder_use_retract": false}], "x_y_axis_movement_speed": null, "z_axis_movement_speed": null, "spiral_vase_mode": false, "layer_height": null, "axis_speed_display_settings": "mm-min"}, "slic3r_pe": {"slicer_type": "slic3r-pe", "version": "unknown", "extruders": [{"retract_length": 6.0, "retract_lift": 0.075, "retract_speed": 40.0, "deretract_speed": 50.0}], "axis_speed_display_units": "mm-sec", "layer_height": 0.15, "spiral_vase": false, "travel_speed": 180}, "other": {"slicer_type": "other", "version": "unknown", "extruders": [{"retract_length": null, "z_hop": null, "retract_speed": null, "deretract_speed": null, "lift_when_retracted": false, "retract_before_move": false, "travel_speed": null, "z_travel_speed": null}], "speed_tolerance": 1, "axis_speed_display_units": "mm-min", "vase_mode": false, "layer_height": null}}, "suppress_snapshot_command_always": true, "auto_detect_position": true, "origin_type": "front_left", "home_x": null, "home_y": null, "home_z": null, "override_octoprint_profile_settings": true, "bed_type": "rectangular", "diameter_xy": 0.0, "width": 240.0, "depth": 240.0, "height": 260.0, "custom_bounding_box": false, "min_x": 0.0, "max_x": 210.0, "min_y": 0.0, "max_y": 210.0, "min_z": 0.0, "max_z": 205.0, "restrict_snapshot_area": false, "snapshot_diameter_xy": 0.0, "snapshot_min_x": 0.0, "snapshot_max_x": 0.0, "snapshot_min_y": 0.0, "snapshot_max_y": 0.0, "snapshot_min_z": 0.0, "snapshot_max_z": 0.0, "auto_position_detection_commands": "", "priming_height": 0.25, "minimum_layer_height": 0.05, "e_axis_default_mode": "absolute", "g90_influences_extruder": "true", "xyz_axes_default_mode": "absolute", "units_default": "millimeters", "axis_speed_display_units": "mm-min", "default_firmware_retractions": false, "default_firmware_retractions_zhop": false, "gocde_axis_compatibility_mode_enabled": true, "home_axis_gcode": "G90; Switch to Absolute XYZ\r\nG28 X Y; Home XY Axis", "num_extruders": 1, "shared_extruder": false, "zero_based_extruder": true, "extruder_offsets": [], "default_extruder": 1, "snapshot_command": "G4 P1"}}`

</details>

mbell90, 2023-01-16:
I'm sorry. I kept forgetting..... :( If I'm able to I'll send the profile I
was using.

tinanewtonart, 2023-08-13:
I am interested in the profile too