FormerLurker/Octolapse #832: Large Cura thumbnails prevent Octolapse from picking up slicer settings

SiliconExarch, 2022-03-25:
#### If this is a feature request describe it here

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.7.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Slice a model with complex surface geometry in Cura with the Create Thumbnail script set to 300x300 pixels
2. Save the file to Disk
3. Upload to OctoPrint making sure Octolapse is enabled
4. Click Print

#### What should have happened?
The analysis should run and show me the snapshot plan.

#### What happened instead?
Octolapse returns the following:
`ERROR - Some slicer settings were missing from your gcode file.  Missing Settings: Retraction Length,Retraction Speed,Deretraction Speed,Z Lift Height,X/Y Travel Speed,Z Travel`

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to the gcode file you were printing when the problem occurred

Any gcode where the thumbnail header extends beyond the first 1000 lines or so...

If I strip the header Octoprint will analyse the file properly.

#### Link to settings.json

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log: (https://gist.github.com/SiliconExarch/7e5f56d2d9660e32fe9ca7e070ab7cdf)

#### Link to octoprint.log

#### Link to contents of Javascript console in the browser

#### Screenshots and/or videos of the problem:

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2022-03-25:
Try adding the settings to the end gcode in this case.  I think that should be suitable.  I may add a setting to increase the number of lines searched, but if it gets too high, it adds a lot of time to the search.  To increase the performance of searching through gcode, I set Octolapse to only search the first and last 1000 lines.  It might not seem like this would add much time, but python is slow, and some files are 100s of megabytes.

Let me know if that helps.

SiliconExarch, 2022-03-29:
> Try adding the settings to the end gcode in this case. I think that should be suitable. I may add a setting to increase the number of lines searched, but if it gets too high, it adds a lot of time [...]

I've just come across another model which produces a large thumbnail after moving the settings to the end gcode a few days ago and sure enough Octolapse picked up the settings just fine :) thanks for the tip.

FormerLurker, 2022-04-26:
Excellent!  I'm going to close this out then.  Thanks for getting back to me, and sorry I didn't get back sooner!