FormerLurker/Octolapse #833: Unable To Enable Custom Preferences (Raspberry Pi HQ)

russoj277, 2022-03-26:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.1

#### Version of OctoPrint

OctoPrint Version: 1.7.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled:

#### What were you doing when the problem occurred

1. I'm attempting to enable Custom Image Preferences for a Rapsberry Pi HQ camera by following the instructions within the help page linked by the blue "?"  at the bottom of the Edit Camera page within Octolapse.

#### What should have happened?
Image preferences should appear and allow changes to be made, such as brightness, contrast, etc.

#### What happened instead?
Brief background: I've been using a Pi HQ camera with Octolapse on a Zero 2 W for a few months now without many issues.  I've recorded dozens of time-lapses with various stabilization and triggers enabled (these options are amazing!), and I've updated octopi.txt to allow for capturing 1080 resolution images.  Also, within octopi.txt, I've always used [camera="raspi"] as I found "auto" and "usb" never worked for me.

1) Per instructions I've found online, "camera=usb" is required within octopi.txt for the custom image preferences to work properly, even when using a Pi camera, so I updated octopi.txt, with "camera="usb" (and set the resolution and frame rate) and after rebooting the Pi I lost my live stream and clicking the "Test Webcam" button fails.  Note: I also updated: [camera_http_webroot="./www"] and [camera_http_options=""].

2) Defaulting back to my usual "camera=raspi" and clicking the "Test Webcam" button results in a green pop-up with positive results, but enabling Custom Image Preferences results in an error message: "Unable to Enable Custom Preferences.  An unexpected error was raised while testing custom image preferences for the 'Raspberry Pi HQ Camera' camera profile."  This behavior was of course to be expected, but I tried it just to ensure my camera was still working.

3) I should also be able to access the control panel manually for mjpegstreamer via http://{IP_ADDRESS_OF_PI}/webcam/control.htm and this isn't working.

4) When using "camera=raspi" (not usb) I can access the live stream with this link as well: http://{IP_ADDRESS_OF_PI}/webcam/?action=stream

5) This could all have something to do with the bcm2835-v4l2 driver, but I don't know how to troubleshoot this.  I did place "bcm2835-v4l2" at the end of the modules file.

6) I ran "sudo apt-get update" and "sudo apt-get dist-upgrade" but didn't see any improvements.

#### Operating System running OctoPrint and Octolapse

OS Name: Windows
Os Version: 10 Pro

#### Printer model & used firmware incl. version

Printer Model: Anycubic Mega X
Printer Firmware Version: 

#### Browser and version of browser, operating system running browser

Browser: Chrome 99.0.4844.51
Browser OS:

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: n/a

#### Link to settings.json

Link to settings.json with all passwords removed: 

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  Where can I find this file?

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/iBaJ75u9

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

RedDogUK, 2023-07-08:
@FormerLurker, I'm running "Octopi (new camera stack)" and using the Camera 3 module.  There are a few things which don't seem to work correctly with this package:

1. The settings per the /boot/camera-stream/libcamera.conf don't seem to take effect as the camera doesn't focus (example attached - I know the image is upside down, I forgot to amend the setting)
> When I view the camera via the browser port (http://octopi.local:8080/), I can pass "http://octopi.local:8080/option?AfMode=2" to the camera and it then auto focuses.

3. When viewing the primary Octolapse screen, the timelapse preview doesn't produce any images (the code does take pictures in the background and compile a video)

4. When editing the settings of the camera, it doesn't allow us to enable "Custom Image Preferences".

When looking at the logs (plugin_octolapse.log), there are a number of recurring bugs in "snapshot.py".  Log attached.

Both the HQ camera and module 3 camera use the libcamera library, so I'm wondering if there is something in common here?

[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/11991608/plugin_octolapse.log)

https://github.com/FormerLurker/Octolapse/assets/40634495/2a8b1e3d-8843-41a2-ab2c-72d6c24e0305

[octoprint-systeminfo-20230708194640.zip](https://github.com/FormerLurker/Octolapse/files/11991903/octoprint-systeminfo-20230708194640.zip)

@russoj277, did you get your HD camera to work?

russoj277, 2023-07-10:
I was able to grab a few time-lapse videos with my camera.  I haven't used it much in the last year.  What free time I have available has been primarily dedicated to post processing my prints and molding/casting rubber parts for a cosplay project.