FormerLurker/Octolapse #835: Issue with Rendering .mp4 files - File not playable/readable

Elitesoulman, 2022-03-30:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.4.1

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.7.3 

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. test mode printing a benchy to test new setup of DSLR timelapse
2. Snapshots are returned successfully
3. Timelapse completes
4. Timelapse renders
5. Timelapse render is not playable in anything other than VLC or after conversion to another file type/resolution.
6. I've attempted multiple settings on both camera and within Octolapse, including resolution and cable/memory card changes on the camera, and FPS, bitrate, rendering core, file type, and file naming scheme changes in Octoprint/Octolapse
7. Snapshot archive is available with all photos 

#### What should have happened?
Timelapse renders .mp4 with settings in rendering profile and is playable after download on most devices including mobile (Android)

#### What happened instead?
Rendered Timelapse is not playable except in VLC

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Prusa Mini+
Printer Firmware Version: Prusa-Firmware-Buddy 4.3.3

#### Browser and version of browser, operating system running browser

Browser: Chrome Version 99.0.4844.84 (Official Build) (64-bit)
Browser OS: Windows

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://drive.google.com/file/d/1oa7lPmPkE_a1xaAxpe4l2fMYLljjFlvP/view?usp=sharing

#### Link to settings.json

Link to settings.json with all passwords removed: https://drive.google.com/file/d/1BFgkON6MLSPvtn_AVxtDYqFkIhJ_s38Y/view?usp=sharing

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log: https://drive.google.com/file/d/1SGbI43K5mRSCqBW-9n2JBlpP74TB3RLF/view?usp=sharing

#### Link to octoprint.log

Link to octoprint.log: https://drive.google.com/file/d/1j6QgGSp5HhULuwcCAl5IGst6oUIAc5Ht/view?usp=sharing

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 

https://drive.google.com/file/d/147qvTi8jm9NFvDR5rbMIIHK5eDLtbclq/view?usp=sharing
https://drive.google.com/file/d/1ETpDLoiuGp_Cmed6VgD_NFcU0xm_ikVk/view?usp=sharing

This is the most recent one with the .zip archive of the snapshots included as well - they seem fine.
Render - https://drive.google.com/file/d/1XYHajCCxq7fHtF_To6DoHN6NpMAae3Nk/view?usp=sharing
ZIP - https://drive.google.com/file/d/12h7JUOw11MkC-56a8UTFR6IMUoCWBrLX/view?usp=sharing

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Elitesoulman, 2022-03-31:
So, in the interest of providing more info and logs, I completely reinstalled Octoprint after a reboot, an I'm having the same issue.

New Logs/Timelapse after refresh
https://drive.google.com/drive/folders/14x63WbMxLh7bZmWtEEhAVcbSqBgskP73

Elitesoulman, 2022-04-02:
I have found a work around currently. After testing and having a successful result using a 1080p webcam, I tried lowering the resolution on the DSLR to S FINE which for my Nikon D50 is 1,504 x 1,000. Perhaps the issue is tied purely to resolution rather than some other setting.

Result is attached:
https://drive.google.com/drive/folders/160y0qCJoV2q8EPvWsePW92QD9eh79zOv

FormerLurker, 2022-04-26:
Not all video players can play all resolutions for mp4 files, unfortunately.  VLC seems to be able to play the ultra high res MP4s.  I've used handbrake to convert these files, though I'll admit it would be better to add some capabilities to the rendering pipeline to handle this.  BTW, 4k resolution should work on most players.