FormerLurker/Octolapse #836: Octolapse Startup Failed Error Starting Timelapse   An unexpected error occurred while starting the timelapse.

ash1300ad, 2022-03-30:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: [Octolapse 0.4.1](https://github.com/FormerLurker/Octolapse/releases/tag/v0.4.1)

#### Version of OctoPrint

OctoPrint Version: _Version 1.7.3

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

1. Select print
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
Should have printed

#### What happened instead?
Octolapse error : Octolapse Startup Failed Error Starting Timelapse   An unexpected error occurred while starting the timelapse.

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: Ender 3 Pro
Printer Firmware Version: _Marlin 2.0

#### Browser and version of browser, operating system running browser

Browser: _safari
Browser OS: _mac os

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: [___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE
](https://gist.github.com/ash1300ad/5b11f656782b16b6181c3b3ddd8dfa28#file-gcode)

#### Link to settings.json

Link to settings.json with all passwords removed: [___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE](https://gist.github.com/ash1300ad/5b11f656782b16b6181c3b3ddd8dfa28/7f36958c6e36fffcd596aa9d09628881d4430a0e)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  [LINK_GOES_HERE](https://gist.github.com/ash1300ad/5b11f656782b16b6181c3b3ddd8dfa28/7f36958c6e36fffcd596aa9d09628881d4430a0e)

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

ash1300ad, 2022-03-30:
> I need to look at the cura settings script you used in detail, but will get back to you soon.
>
> I have been unable to deal with GitHub issues for the last couple months due to work/life, but things are finally getting back to normal. I do have a lot of catching up to do though!

Hello, 

I was able to fix the issue. There was an error and apparently it corrupted the gcode I was printing from. I sliced the STL again and was able to print from that one without issue

Thanks

FormerLurker, 2022-04-26:
Excellent, going to close this out, thanks for reporting back!