FormerLurker/Octolapse #837: Extruder retracting filament back to the extruder gear.

Viper5343, 2022-04-02:
#### Version of Octolapse
Octolapse Version: Octolapse 0.4.1

#### Version of OctoPrint
OctoPrint Version: OctoPrint 1.7.3 

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred
1. Printing via octoprint.
2. Tried with start Gcode at start, Also tried with it at the end.
3. Same thing happed with an Arc weld file and standard Gcode.
4. G90/G91 overrides relative extruder mode is checked.

#### What should have happened?
The print should have completed.

#### What happened instead?
It retracted almost if not all the filament back to the extruder gear around the start of the 2nd or 3rd layer.

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi
Os Version: OctoPi 0.18.0

#### Printer model & used firmware incl. version
Printer Model: Cr-10s
Printer Firmware Version: Marlin TH3D UFW 2.36

#### Browser and version of browser, operating system running browser
Browser: Google Chrome Version 99.0.4844.84
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: https://drive.google.com/file/d/1eDqApkvcJh8vBmUVinv9i4dqiZhJVKCA/view?usp=sharing

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log: https://drive.google.com/file/d/12Epm8yub3Hir3V0awdn5BeSoM7kFmnE5/view?usp=sharing

#### Link to octoprint.log
Link to octoprint.log:  https://drive.google.com/file/d/1eDqApkvcJh8vBmUVinv9i4dqiZhJVKCA/view?usp=sharing

Viper5343, 2022-04-26:
Acording to the printer display screen it is currently runing TH3Ds merlin 2.36.

FormerLurker, 2022-04-26:
Looks like marlin 1.x:

[image]

This is from their [github repo](https://github.com/th3dstudio/TH3D-Unified-U1.R2), so G90/G91 does not change the extrusion mode.

Viper5343, 2022-04-26:
Ok, thank you I will try again with that option unchecked and post an update. Thank you for your time.

FormerLurker, 2022-04-26:
Of course!  Looking forward to your response.