FormerLurker/Octolapse #839: Installation problems

ilmelo92, 2022-04-13:
Hello, I have been trying to install octolapse for days, unfortunately, always with negative results

Version of Octolapse  

Octolapse Version: I try to install the v0.4.1

#### Version of OctoPrint

OctoPrint Version:  v1.7.3

#### Operating System running OctoPrint and Octolapse

OS Name: QTS (Qnap Systema)
Os Version: 5.0.0.1986

#### Printer model & used firmware incl. version

Printer Model: Artillery Sidewinder X2
Printer Firmware Version:  v4.0.3

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows 11

#### Link to plugin_octolapse.log

https://pastebin.com/YTyZffKp

#### Link to octoprint.log

https://pastebin.com/HnTfz0d9

I hope I have filled in everything correctly, please forgive me if this is not the case
I also apologize for my bad english

pretty much every time i try to install this great plugin i get this error

https://pastebin.com/J5itmeWz

never completing the installation

Hello and thank you very much for your work

FormerLurker, 2022-04-26:
It looks like gcc is missing (the c++ compiler).  Can you give me more details about the QTS OS?  I'm unfamiliar with that one.

ilmelo92, 2022-05-04:
hello dear, thank you very much for your reply
what information could you find useful? I will do everything to let you have them

I add that however actually I am not even finding 1 c ++ compiler for qnap so far = (

FormerLurker, 2022-05-04:
Maybe try:

```sudo apt install build-essential```

Not sure if that is the proper command for qnap.  I have 0 experience with that :)

ppastur, 2022-09-18:
@FormerLurker , QNAP is a NAS and has a very customised (cut down and controlled) version of UBUNTU.  @ilmelo92 you might have better luck running it in a Docker container or Pi. Apologies if I'm overstepping.