FormerLurker/Octolapse #840: Problems installing plugin

SHoover80, 2022-04-17:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Version 0.4.1

#### Version of OctoPrint

OctoPrint Version: Version 1.7.3 

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Not sure

#### What were you doing when the problem occurred

1. I installed the plugin 
2. I restarted Octoprint 
3. 

#### What should have happened?
I should have been able to start adjusting settings for Octolapse

#### What happened instead?
Octolapse was enable to load the current settings.
Status: error. Error. INTERNAL SERVER ERROR

#### Operating System running OctoPrint and Octolapse

OS Name: Windows 11 Pro 
Os Version: 21H2 OS Build 22000.613

#### Printer model & used firmware incl. version

Printer Model: Creality Ender-5 Pro
Printer Firmware Version: Marlin V1.2.1

#### Browser and version of browser, operating system running browser

Browser: Google Chrome
Browser OS: Version 100.0.4896.88 (Official Build) (64-bit)

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: N/A

#### Link to settings.json

Link to settings.json with all passwords removed: All settings are default, Brand new printer just trying to setup Octoprint and Octolapse. 

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:N/A

#### Link to octoprint.log

Link to octoprint.log:  N/A

#### Link to contents of Javascript console in the browser

Link to javascript console output: N/A

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://imgur.com/9zNTaXp

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).
[image]

FormerLurker, 2022-04-26:
This might be an issue that's resolved in the dev build.  What version of python do you have installed?  If it's the latest, you can try to install from the 'devel' branch using this url (in the from url... area in the plugin manager):  https://github.com/FormerLurker/Octolapse/archive/refs/heads/devel.zip

SHoover80, 2022-04-27:
Thank you for the response, I did get that installed and it appears to be working so far. I still got to set it up but I didn't get any errors after installing/restarting so it's looking good.

FormerLurker, 2022-04-27:
Thanks @SHoover80 !  Let me know when you have a working timelapse please :)

SHoover80, 2022-04-27:
Yup @FormerLurker, Everything is working great now. Thank you.

FormerLurker, 2022-04-27:
Hooray!!  Ok, will close this.  Enjoy!

closed 2022-04-27 by FormerLurker