FormerLurker/Octolapse #842: Will not install 'gcc' failed

Brodieman, 2022-04-18:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 1.7.3

#### Version of OctoPrint

OctoPrint Version: 0.4.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. Installing
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
Should have installed

#### What happened instead?
Does not install, with this error:
error: command 'gcc' failed: No such file or directory
----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-70q1wtpu/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-70q1wtpu/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-zpd6yg8w/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/Octolapse Check the logs for full command output.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
----------------------------------

#### Operating System running OctoPrint and Octolapse

OS Name: Android
Os Version: 12

#### Printer model & used firmware incl. version

Printer Model: Ender 3 S1
Printer Firmware Version: 1.0.5

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2022-04-26:
Guessing you are running Octo4a?  In the app, in the settings menu, there is an option called "install plugin extras".  Try that option and let me know if it helps.

Brodieman, 2022-04-28:
That did it, thanks

Brodieman, 2022-04-28:
Except now when I try to print from Cura to OctoPrint I ge this
[image]

FormerLurker, 2022-04-28:
Read the [getting started](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started) guide, and pay attention to the [Automatic Slicer Configuration](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started#use-automatic-slicer-configuration) section.  Getting [cura to work with this](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Automatic-Slicer-Configuration) requires a couple simple steps.

FormerLurker, 2022-04-28:
Not a problem.  Let me know if you get it working (I suggest starting in test mode, then disabling test mode once you've verified things are working), and I'll close the issue. If not, back to the drawing board :)